OpenEPaperLink/OpenEPaperLink #212: Hanshow 7.5" support

nico-arnold, 2024-01-31:
Hey everybody,

thanks for your great work here! 
Is there perhaps already support (planned) for the Hanshow 7.5" tags? I have one here that I got from a local supermarket and cut it open a few days ago, so the pins are exposed. Looks much better than what I saw at the smaller tags, as they use an FPC connector here instead of heat-bonding the display to the board. Additionally, they seem to have Li-Ion batteries inside (no recharging port, though).
If I can help you somehow or test something here, please let me know!

best wishes,
Nico

jjwbruijn, 2024-02-04:
Since there aren't a lot of those tags out there, let alone many in hands of the core dev team, I guess you're going to have to write support yourself :) We take pull requests!

closed 2024-02-04 by jjwbruijn

nico-arnold, 2024-02-04:
Ah, okay, too bad. But if they use the same principles on all their devices it might be possible. I'll have a look at the code when I have some spare time, maybe I can do it by myself (: Thanks for your answer!