OpenEPaperLink/OpenEPaperLink #214: Problems with MiniAP and SD-Card

seeers, 2024-02-01:
I soldered an SD card slot to the MiniAP and adapted the Platformio.ini accordingly:

board_build.partitions = esp32_sdcard.csv
and added the following buildflags:
	-D HAS_SDCARD
	-D USE_SOFTSPI
	-D SD_CARD_SS=38
	-D SD_CARD_CLK=40
	-D SD_CARD_MISO=37
	-D SD_CARD_MOSI=39

Changed 'VSPI' to 'HSPI' in storage.cpp, erased the Flash, build and uploaded the compiled FW. 

I can connect to the APs wifi, unfortunately the status is initializing... 
Sometimes the AP reboots / crashes when I refresh the page http://192.168.0.4 with F5 (Windows sound for new hardware / COM port found) 
The segmented display shows IP addr., then nothing follows. 

Access to the SD card works via http://192.168.4.1/edit

Have I made a fundamental mistake or could it be a bug? 
Does anyone else use an ESP32-S2 with Segmented Display as a radio and an SD card? 

[image]
Serial output:
```
| AP Info - type F0       |
| Ch   |             0x0B |
| Power|               0A |
| MAC  | 840B2D9150650131 |
| Ver  |           0x0017 |
[WiFi-event 12]
[WiFi-event 14]
>AP Segmented Data 840B2D9150650131
queue item added, first in line, total 1 elements
ACK>
XFC>Waiting for valid time from NTP-server
[WiFi-event 14] 
>AP Segmented Data 840B2D9150650131
queue item added, total 2 elements
>AP Segmented Data 840B2D9150650131
queue item added, total 3 elements
>SDA 840B2D9150650131
queue item added, total 4 elements
AP FW version 0017 - AP_FW_Segmented_UK.bin found in FW pack
Waiting for valid time from NTP-server
Waiting for valid time from NTP-server
Waiting for valid time from NTP-server
Waiting for valid time from NTP-server
pingACK> ok, 4ms
ADR>pingACK> ok, 4ms
```

Log.txt:
```
           00:00:03 No connection information saved
           00:00:03 Starting configuration AP, ssid OpenEPaperLink
           00:00:03 Reboot. Reason: Power-on
           00:00:03 No connection information saved
           00:00:03 Starting configuration AP, ssid OpenEPaperLink
           00:00:03 Reboot. Reason: Power-on
           00:00:03 No connection information saved
           00:00:03 Starting configuration AP, ssid OpenEPaperLink
           00:00:03 Reboot. Reason: Power-on
           00:02:00 No connection information saved
           00:02:00 Starting configuration AP, ssid OpenEPaperLink
           00:02:00 Reboot. Reason: Task Watchdog
           00:07:17 No connection information saved
           00:07:17 Starting configuration AP, ssid OpenEPaperLink
           00:07:17 Reboot. Reason: Panic
           00:00:03 No connection information saved
           00:00:03 Starting configuration AP, ssid OpenEPaperLink
           00:00:03 Reboot. Reason: Power-on
```

jonasniesner, 2024-02-17:
To be honest the SD card code is not really tested because no currently supported APs support it. Why do you need an SD card?

seeers, 2024-02-18:
I use relatively large vlw fonts with icons, so I didn't have much space left on the filesystem. I also suspected that the memory area had been corrupted. (little space and many updates) 
So I wanted to try out the option with the SD card. 
In the meantime I have changed to a ttf font, now there is more space on a new esp32-s2. 

Nevertheless I find the SD option interesting to save write cycles of the internal flash.

jonasniesner, 2024-03-02:
I am closing this for now since we have no capacity to work on this at the moment and no "official" AP uses it at the moment. Feel free to work on it yourself though.

closed 2024-03-02 by jonasniesner