OpenEPaperLink/OpenEPaperLink #216: hanshow nebular problem

nikkisstuff, 2024-02-03:
I have two hanshow nebular 3.5" displays both the same model but one or them the background colours are inverted 

[image]

nethomas1968, 2024-02-04:
I've heard that there are 2 chipsets used in these devices. Is it possible that you have one of each.
I suppose you might need to open up the back to get to see the little CPU board to look at the chip markings.

nethomas1968, 2024-02-04:
Also, which of the Firmware directories in this repo have you used to get that far ? I've had little success with the ARM_Tag_FW/OpenEPapaerLink/TLSR but I can't get it to use the whole screen area.
What are you doing right that I'm not ?

nikkisstuff, 2024-02-05:
This is the file i used https://github.com/jjwbruijn/OpenEPaperLink/blob/master/binaries/Tag/TLSR_HS_35_ALPHA.bin

I did open the tags and the board look the same. the chip is identical just branded "Hanshow"

nikkisstuff, 2024-02-05:
Also I am running it via OpenEPaperLink-PyStation with a cc2531

closed 2024-02-05 by nikkisstuff