OpenEPaperLink/OpenEPaperLink #218: Tags stuck on 'Updating' when going from 36 (0x24) to 38 (0x26)

ThinkPadNL, 2024-02-04:
**Describe the bug**
M2 tags are stuck on black screen 'Updating!' and do not come back online when updating from 0x24 to 0x26

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'advanced settings'
2. Click on 'auto update'
3. Tag changes to 'updating!'
4. Tag is not coming back online again, bricked?

**Expected behavior**
Update succeeding and tag coming back online

**Additional context**
Looks same issue as #184 but my tags were already on v26

jonasniesner, 2024-02-05:
Hi, yea, you will most likely have to reflash with a jig. The bricking protection is only active starting with the firmware you where trying to update to. Try removing the batteries and inserting them back after shorting the pins first.

nlimper, 2024-02-08:
The bug that would cause corrupted transfers is fixed in version 0x25, so in 0x24 it still could go wrong. 
If they are stuck on 'Updating', using a jig to reflash will be the only option sadly.

closed 2024-02-08 by nlimper