OpenEPaperLink/OpenEPaperLink #231: Tag DB resets every few weeks

foorschtbar, 2024-02-17:
**Describe the bug**
Every few weeks my five displays doenst refresh anymore. Checking the AP shows that all tags are resetted to static image. happens every 2-4 weeks. @metaln00b hat the same problem.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup tags
2. wait 2 weeks
3. all tag settings deleted

**Expected behavior**
n/a

**Screenshots**
n/a

**Additional context**
n/a

nlimper, 2024-02-17:
Did you use the latest release (2.07b), or the most recent version at the master repo? Between those, much has happened regarding stability (and probably, some other bugs introduced instead ;-) .

Metaln00b, 2024-02-17:
> Did you use the latest release (2.07b), or the most recent version at the master repo? Between those, much has happened regarding stability (and probably, some other bugs introduced instead ;-) . 

Already installed 🫣

foorschtbar, 2024-02-17:
Same here. Also on 2.07b

jum0n, 2024-02-17:
I switched back to 2.5 since 2.07b does not work on my gateway.   I see we can download the tagdb, is there a way to restore it?

nlimper, 2024-02-17:
I'm sorry, it doesn't make sense trying to fix bugs or provide support for older versions that might be already fixed in newer versions. Feel free to reopen this issue if it's still there after updating it.

Yes, it's possible to restore the DB with a button, from version 2.07b. Also, in the latest version on master, you can even restore a database copy that is always kept in the localstorage, so even without making a backup, you can restore your tag content.
If 2.07b doesn't work on your gateway, that's something to look into. It's good to investigate (using the latest version on master, or wait until the next release which I expect to be somewhere the coming week).

closed 2024-02-17 by nlimper

foorschtbar, 2024-02-17:
Your comment make no sense to me. I'm on the latest official version and I open the the ticket. There is a second person whit the same problem also on 2.07b.

I'm not here to get a instant fix, but it makes sense for me to report a problem.

Should I now wait or switch to master or could I provide additional information that helps?

nlimper, 2024-02-17:
Best is to wait, because probably it's already fixed in the upcoming release, which has a lot of stability improvements. 
If it's still there, feel free to reopen this issue.

Metaln00b, 2024-02-17:
> Best is to wait, because probably it's already fixed in the upcoming release, which has a lot of stability improvements. 
> If it's still there, feel free to reopen this issue. 
> 

Thank you 👍

foorschtbar, 2024-02-17:
> Best is to wait, because probably it's already fixed in the upcoming release, which has a lot of stability improvements. 
> 
> If it's still there, feel free to reopen this issue. 
> 
> 

Makes sense. Thank u 🙏🏻