OpenEPaperLink/OpenEPaperLink #232: [Support] ESP32-S3-DevKitC-1-N8R2 board support

beralt85, 2024-02-17:
**Feature: support for ESP32-S3-DevKitC-1-N8R2**
See datasheet: https://www.mouser.de/datasheet/2/891/esp32_s3_wroom_1_wroom_1u_datasheet_en-2930317.pdf

**Is your feature request related to a problem? Please describe.**
Due to the variety of different platforms for the ESP32-S3-DevKitC, not all environments work out of the box.
Sometimes, if one orders one board one might get sent another board and only realize afterwards (as in my case, I ordered N16R8 version for AP_and_flasher, but got send the N8R2)

**Describe the solution you'd like**
Add more boards to the `platformio.ini` file.

**Additional context**
Here is a the working environment for the ESP32-S3-DevKitC-1-N8R2, using the default 8MB partition layout.
Some tweaking was required to get PSRAM to work:

```
[env:OpenEPaperLink_AP_and_Flasher-one-at-atime]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-s3-devkitc-1
board_build.partitions = default_8MB.csv
build_unflags =
	-std=gnu++11
	-D ARDUINO_USB_MODE=1
	-D CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC=y
build_flags = 
	-std=gnu++17
	${env.build_flags}
	-D HAS_EXT_FLASHER
	-D ARDUINO_USB_MODE=0
	-D CONFIG_ESP32S3_SPIRAM_SUPPORT=1
	-D CONFIG_SPIRAM_USE_MALLOC=1
	-D HAS_USB
	-D HAS_RGB_LED
	-D BOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue # was suggested
 	-D CONFIG_SPIRAM_CACHE_WORKAROUND # had this in old config
	-D CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
	-D POWER_RAMPING
	-D POWER_HIGH_SIDE_DRIVER
	-D FLASHER_AP_SS=4
	-D FLASHER_AP_CLK=5
	-D FLASHER_AP_MOSI=7
	-D FLASHER_AP_MISO=6
	-D FLASHER_AP_RESET=15
	-D FLASHER_AP_POWER={0} 
	-D FLASHER_AP_TXD=18 #was 16, because I soldered stupidly
	-D FLASHER_AP_RXD=16 #was 18, because I soldered stupidly
	-D FLASHER_AP_TEST=17
	-D FLASHER_EXT_SS=40
	-D FLASHER_EXT_CLK=41
	-D FLASHER_EXT_MOSI=2
	-D FLASHER_EXT_MISO=42
	-D FLASHER_EXT_RESET=1
	-D FLASHER_EXT_POWER={8}
	-D FLASHER_EXT_TXD=38
	-D FLASHER_EXT_RXD=39
	-D FLASHER_EXT_TEST=47
	-D FLASHER_ALT_SS=3
	-D FLASHER_ALT_CLK=46
	-D FLASHER_ALT_MOSI=10
	-D FLASHER_ALT_MISO=9
	-D FLASHER_ALT_RESET=11
	-D FLASHER_ALT_POWER={-1}
	-D FLASHER_ALT_TXD=12
	-D FLASHER_ALT_RXD=14
	-D FLASHER_ALT_TEST=13
	-D FLASHER_LED=21
	-D FLASHER_RGB_LED=48
build_src_filter = 
	+<*>-<espflasher.cpp>
board_build.flash_mode=qio #qio important for this board
board_build.arduino.memory_type = qio_qspi #This line is important for integrated quad psram
board_build.psram_type=qspi_qpi # needed to be changed, since the psram is quad spi
board_upload.maximum_size = 4194304 # changed to 4MB, not quite sure whether this is the max size of the firmware (which would be still to large) or the whole image
board_upload.maximum_ram_size = 327680
board_upload.flash_size = 8MB  #changed to 8
```
There are some things that I don't really understand (see comments), where maybe some things could be optimized. The same is for a more optimal partition layout.
Also note that with respect to the standard AP_and_Flasher config, I switched  (because I was confused) the UART TX/RX pins for the primary radio. In any case, maybe you want to add an environment for this board.

I know, I should probably do this as a Pull request, but have never contributed to a public repo, so I did it like this. If you are interested in integrating this, I  can probably also go the official pull request way.

jonasniesner, 2024-02-17:
It looks like you just built a one off AP. In that case I would advise that you keep the config in your own fork. We stopped accepting one off board configs in the main repo. So if you need OTA support from the main repo, it is advised to get a supported board. 
We had to do this change because testing would be even more of a nightmare with more different boards and boards with lower ram and flash. Please don't let this discourse you from tinkering on your own fork.

closed 2024-02-17 by jonasniesner

beralt85, 2024-02-18:
Hey Jonas,
no problem :) 
I was planning to go with a supported board, but I got the wrong ones and was too late to complain ;) And yes, I was hoping for OTA ^^
Thanks for all your (and ATCs and dmitrys) work, this is amazing upcycling.