OpenEPaperLink/OpenEPaperLink #233: Mini-AP loses WiFi

dl6lr, 2024-02-19:
After having run OpenEPaperLink for almost a year, I updated to 2.07b. Things went smooth, but now the AP loses its Wifi connection very often. Sometimes it runs for only one hour, then loses connection and does not reconnect, but spans up its own SSID for configuration.

This can't be reproduced, but occurs at random intervals.

Log of last events:
```
2024-02-18 22:17:04 http imageUpload 026D256A3B1B.jpg
2024-02-18 22:21:59 WiFi connection lost. Attempting to reconnect.
2024-02-18 22:22:15 Unable to connect to WiFi
2024-02-18 22:22:15 Starting configuration AP, ssid OpenEPaperLink
2024-02-18 22:27:15 Attempting to reconnect to WiFi.
2024-02-18 22:27:16 Connecting to WiFi...
2024-02-18 22:27:16 Connected!
2024-02-18 23:17:03 http imageUpload 026D256A3B1B.jpg
2024-02-19 00:17:04 http imageUpload 026D256A3B1B.jpg
2024-02-19 01:17:04 http imageUpload 026D256A3B1B.jpg
2024-02-19 01:33:34 WiFi connection lost. Attempting to reconnect.
2024-02-19 01:33:49 Unable to connect to WiFi
2024-02-19 01:33:49 Starting configuration AP, ssid OpenEPaperLink
2024-02-19 04:00:00 Nightly reboot
2024-02-19 04:00:04 Reboot. Reason: Software
2024-02-19 04:00:05 Connecting to WiFi...
2024-02-19 04:00:06 Connected!
2024-02-19 04:00:09 840B2D90EE5C0131 Booting
2024-02-19 04:17:04 http imageUpload 026D256A3B1B.jpg
2024-02-19 04:22:43 WiFi connection lost. Attempting to reconnect.
2024-02-19 04:22:58 Unable to connect to WiFi
2024-02-19 04:22:58 Starting configuration AP, ssid OpenEPaperLink
2024-02-19 04:27:58 Attempting to reconnect to WiFi.
2024-02-19 04:27:59 Connecting to WiFi...
2024-02-19 04:27:59 Connected!
2024-02-19 05:17:04 http imageUpload 026D256A3B1B.jpg
2024-02-19 06:17:04 http imageUpload 026D256A3B1B.jpg
2024-02-19 06:25:02 http imageUpload 026d29c13b16.jpg
2024-02-19 07:04:04 WiFi connection lost. Attempting to reconnect.
2024-02-19 07:04:19 Unable to connect to WiFi
2024-02-19 07:04:20 Starting configuration AP, ssid OpenEPaperLink
2024-02-19 07:50:33 0000026D6C993B1A Network scan
           01:00:03 Reboot. Reason: Power-on
           01:00:04 Connecting to WiFi...
           01:00:05 Connected!
2024-02-19 08:21:02 840B2D90EE5C0131 Booting
2024-02-19 08:44:50 http imageUpload 026d29c13b16.jpg
2024-02-19 09:17:03 http imageUpload 026D256A3B1B.jpg
2024-02-19 10:17:04 http imageUpload 026D256A3B1B.jpg
2024-02-19 11:04:23 Reboot. Reason: Task Watchdog
2024-02-19 11:04:24 Connecting to WiFi...
2024-02-19 11:04:39 Unable to connect to WiFi
2024-02-19 11:04:39 Starting configuration AP, ssid OpenEPaperLink
2024-02-19 11:04:42 840B2D90EE5C0131 Booting
           01:00:03 Reboot. Reason: Power-on
           01:00:04 Connecting to WiFi...
           01:00:05 Connected!
           01:00:08 840B2D90EE5C0131 Booting
2024-02-19 11:09:07 Reboot request by user
2024-02-19 11:09:12 Reboot. Reason: Software
2024-02-19 11:09:12 Connecting to WiFi...
2024-02-19 11:09:14 Connected!
2024-02-19 11:09:17 840B2D90EE5C0131 Booting
2024-02-19 11:17:03 http imageUpload 026D256A3B1B.jpg
2024-02-19 12:17:04 http imageUpload 026D256A3B1B.jpg
2024-02-19 12:22:46 WiFi connection lost. Attempting to reconnect.
2024-02-19 12:23:01 Unable to connect to WiFi
2024-02-19 12:23:01 Starting configuration AP, ssid OpenEPaperLink
           01:00:03 Reboot. Reason: Power-on
           01:00:04 Connecting to WiFi...
           01:00:05 Connected!
2024-02-19 14:49:47 840B2D90EE5C0131 Booting
```

Any hints?

dl6lr, 2024-02-22:
My AP has become extremely unstable with 2.07b. It now has lost all its configured tags, they show up without alias, even the segmented tag of the AP was lost. Furthermore, I see some odd files in the littlefs, in the root there are two .raw files that I expected to be in /current, in /current one raw file of one tag is an old one, where the tag shows a newer one uploaded by the http api.
I downgraded to 2.05 and this is more or less stable again. The WiFi connection is retained but I see some Panic-Reboots. Have to connect to a computer and catch the output...

jonas1972, 2024-02-24:
I have similar behaviour. My system always boots and workshop for a short while, bur then becomes unresponsive and mon-working. I can’t reach the AP web server Using the IP address when that happens. Nothing looks strange on the display. Tris started arter upgrade to 2.07b but I have downgraded to 2.05 and unfortenately I still have problems.

dl6lr, 2024-02-24:
I updated to 2.07b again and tried to gather a serial log, but the AP stopped working without any notice. Now it is flashing yellow(?)/yellow/red and the blue ESP-LED sometimes blinks.

The famous last words on the debug:
```
sleeping for 5 more minutes
>SDA 840B2D90EE5C0131 NOP
ACK>

ADR><ADR 0000026D29C13B16
ADR><ADR 840B2D90EE5C0131
sleeping for 5 more minutes
>SDA 0000021C37FE341C NOP
ACK>

ADR><ADR 0000021C37FE341C
ADR><ADR 0000026D6C993B1A
```

Here is another log where it had a hard time connecting to Wifi:
```
[WiFi-event 0]

[WiFi-event 2]

Stored ssid: FRUEHJAHRSPUTZ

[WiFi-event 3]

[WiFi-event 0]

[WiFi-event 2]

Connecting to WiFi...

[WiFi-event 5]

WiFi lost connection. Reason: 202

[WiFi-event 5]

WiFi lost connection. Reason: 15

Waiting for valid time from NTP-server

[WiFi-event 5]

WiFi lost connection. Reason: 15

[WiFi-event 5]

WiFi lost connection. Reason: 15

Waiting for valid time from NTP-server

[WiFi-event 5]

WiFi lost connection. Reason: 15

!Unable to connect to WiFi

[WiFi-event 5]

WiFi lost connection. Reason: 15

Starting configuration AP, ssid: OpenEPaperLink

[WiFi-event 5]

WiFi lost connection. Reason: 8

[WiFi-event 3]

[WiFi-event 0]

[WiFi-event 10]

[WiFi-event 11]

[WiFi-event 10]

Connect to it, visit http://192.168.4.1/setup

reading DB from /current/tagDB.json

loadDB took 106ms

pingFree heap: 79736 allocatable: 57332 stack: 13300
Waiting for valid time from NTP-server

 failed

Resetting tag

DB saved 357ms

Setting up EPD

RES>

RDY>

AP did reset, resending pending

pingACK> ok, 5ms
ACK>

ACK>TYP>

VER>

MAC>

ZCH>

ZPW>

PEN>

NOP>

ADR><ADR 840B2D90EE5C0131

| AP Info - type F0       |
| Ch   |             0x0B |
| Power|               0A |
| MAC  | 840B2D90EE5C0131 |
| Ver  |           0x0017 |
>AP Segmented Data 840B2D90EE5C0131
ACK>

```

Then after a power cycle:

```
[WiFi-event 0]

[WiFi-event 2]

Stored ssid: FRUEHJAHRSPUTZ

[WiFi-event 3]

[WiFi-event 0]

[WiFi-event 2]

Connecting to WiFi...

[WiFi-event 4] Connected to access point

[WiFi-event 7] Obtained IP address: 192.168.178.30

Connected!

reading DB from /current/tagDB.json

loadDB took 381ms

pingFree heap: 76868 allocatable: 53236 stack: 13348
 failed

```

nlimper, 2024-02-28:
Please update to version 2.5-beta3 and see it it has improved. A lot has changed in the latest versions regarding stability.
Also, in case of WiFi problems, make sure it's not caused by your WiFi access point. The ESP32-S3 has a very basic WiFi stack, and lacks the extended possibilities of sophisticated WiFi routers.
I'm closing this one, but feel free to re-open if needed.

closed 2024-02-28 by nlimper