OpenEPaperLink/OpenEPaperLink #234: Missing Miso Pad on SLT-EM1043

tjspacelord, 2024-02-20:
Hello,

today I was looking to make a mini AP and stumbled upon a little Problem.
I was opening the Segmented Display and there were 11 pads and not 12 like in all the Documentation. I saw that I have an SLT-EM1043 and not a SLT-EM104, that model seems to have only 11 pads. Does somebody knows if it is possible to use this tag for a Mini AP. If yes what would be the pinout for this model, assuming the pinout is identical to the SLT-EM104 tag, then the Miso Pad is missing. And it seems like an important connection.
Here is a picture of the Segmented Display, where the pad is missing.
[image]

Thank you all in advance (This Project is awesome)

jjwbruijn, 2024-02-21:
Interesting! This is a nRF-based segmented tag, which uses a different pinout. While there is support for an nRF based AP, I advise you to build something using an ESP32 C6 or zbs243 based AP instead, as they’re well tested and work with current releases of the AP firmware. I recall there were some range-issues with the nRF AP too…

jjwbruijn, 2024-02-21:
Also, if you want to trade one of these tags for a well supported one, send me a message on discord. I’d like to bring support for segmented tags one day, including this nRF one (should be easy as we already have a universal firmware for the nRF based tags)

closed 2024-02-21 by jjwbruijn