OpenEPaperLink/OpenEPaperLink #235: Flashing C6 failed

hmax42, 2024-02-23:
First my epaper tag with the weatherforecast wouldn't update anymore, so i thought maybe the service died or anything else.
So I did an AP update from 2.01b to 2.07b, whichworked.
Next, the C6 update, which failed

```
env:                ESP32_S3_16_8_YELLOW_AP
build date:         2024-01-09 21:42
esp32 version:      2.07b
filesystem version: 2.07b
psram size:         8383751
flash size:         16777216
--------------------------
Flashing ESP32-C6...
Stopping AP service
C6 flash starting
downloading /bootloader.bin
downloading /partition-table.bin
downloading /OpenEPaperLink_esp32_C6.bin
Download failed, 125952 bytes
Retry 0
downloading /OpenEPaperLink_esp32_C6.bin
Download failed, 76800 bytes
Retry 1
downloading /OpenEPaperLink_esp32_C6.bin
Download failed, 94208 bytes
Retry 2
downloading /OpenEPaperLink_esp32_C6.bin
Download failed, 103424 bytes
Retry 3
downloading /OpenEPaperLink_esp32_C6.bin
Download failed, 94208 bytes
Retry 4
downloading /OpenEPaperLink_esp32_C6.bin
Download failed, 103424 bytes
Retry 5
downloading /OpenEPaperLink_esp32_C6.bin
Download failed, 51200 bytes
Retry 6
downloading /OpenEPaperLink_esp32_C6.bin
Download failed, 125952 bytes
Retry 7
downloading /OpenEPaperLink_esp32_C6.bin
Download failed, 110592 bytes
Retry 8
downloading /OpenEPaperLink_esp32_C6.bin
Download failed, 76800 bytes
Retry 9
C6 flash end
Flashing failed. :-(
```
i did reboot the AP before the C6 update

any advice what to do ?

hmax42, 2024-02-23:
after that i did a rollback, which somehow failed; the webpage didn't load
so i reflashed via the webinterface and am now on 2.05
the weatherforecast still doesn't work

nlimper, 2024-02-23:
Please use the latest firmware version (currently 2.07b).
If your access point shows 'online' as a status, the C6 doesn't need to be updated.

So, that leaves 'weatherforecast wouldn't update'. For that, first check if your tag is working. In the AP, you see the 'last seen' time running, that's the time since the last check in. Normally, the tag checks in every 1-10 minutes. If your AP has been unavailable for a while, this takes longer, eventuelly even just once a day. If it doesn't check in, you can reboot a tag by power cycling it, but note that's it not as easy as just removing and putting back the batteries. Here is the correct procedure: https://github.com/jjwbruijn/OpenEPaperLink/blob/master/Hardware/OpenEPaperLink%20Mini%20AP/Getting%20Started.md#adding-tags
If the tag doesn't start up even after this procedure, your batteries might be empty.

When you have the tag checked in correctly, you can assign a content type to it, like the weather forecast. If the weather forecast doesn't work, but for example a 'count hours' does work, check your internet connection.

hmax42, 2024-02-23:
powercycled incl cap discharge
i think the problem is that the name cannot be resolved to lat/lon/timezone anymore ?
these 3 fields are null, and i tried now different string which lead all (if googled) to the correct town

nlimper, 2024-02-23:
Ah, if the name cannot be resolved, that's the problem! Also, because of the C6 firmware download failed, you should take a look at your WiFi configuration. If possible, try a different WiFi connection to see if that helps. If you're using a Unify WiFi Access Point, Disable 'BSS Transition', because enabling seems to cause connection problems. The WiFi stack of the esp32 is very simple, so sophisticated WiFi protocol enhancements might confuse it.

hmax42, 2024-02-23:
I have a FritzBox access point, i don't know what BSS Transition is, and its firmware wasn't updated since october.

the  download/update of the regular firmware works flawlessly, but the  C6 fiemware alwys fails at some point

what is the address used for the location service? i could test that url from the commandline

hmax42, 2024-02-23:
if i check the c6 serial
```

I (106) esp_image: segment 0: paddr=00010020 vaddr=42020020 size=0b150h ( 45392) map
I (119) esp_image: segment 1: paddr=0001b178 vaddr=40800000 size=04ea0h ( 20128) load
I (126) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=1f2e0h (127712) map
I (144) esp_image: segment 3: paddr=0003f308 vaddr=40804ea0 size=0a430h ( 42032) load
E (150) esp_image: Checksum failed. Calculated 0xff read 0x66
E (150) boot: Factory app partition is not bootable
E (153) boot: No bootable app partitions in the partition table
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x3 (LP_SW_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001974a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xec8
load:0x4086e610,len:0x2ebc
load:0x40875728,len:0x1940
SHA-256 comparison failed:
Calculated: 8bb43836d0bd4a3b86bfbe09db2ec22b5d50360b10bcb10ade6e4300e0fb4f70
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4086c410
I (41) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (41) boot: compile time Oct  1 2023 20:23:29

```

where can i download and flash the c6 manually ?

nlimper, 2024-02-23:
You should not have flashed the C6 again, because that was working correctly...
The C6 now is incompletely flashed, because you flashed it with an incomplete file (the failed downloads).
Now, you can manually download the files needed: download all four files from https://github.com/jjwbruijn/OpenEPaperLink/tree/master/binaries/ESP32-C6 (choose 'download', *don't* use right click/save as, as that downloads the html page), and put the in the root of the AP filesystem. Then click the update C6 button again (with 'download latest' unchecked).

The location service runs in your browser, when you enter the location and get the drop down. See the 'network' tab in the developer tools of your browser. It's somewhere at open-meteo.com . No drop-down? Then you need to update the file system. Only use the latest (2.07b currently)

closed 2024-02-23 by hmax42