OpenEPaperLink/OpenEPaperLink #236: [Question] Segmented AP is a 'AP without radio' after flashing 2.5-beta?

Calimerorulez, 2024-02-25:
Hi,

Thank you for the constant improvements of this project.

Is it correct that a segmented AP now is a 'AP without radio' in 2.5-beta?

The WebUI is showing that on one of my APs.

If that is so, what's the use of that kind of AP?

jonasniesner, 2024-02-25:
No, that is not correct. Did this AP show online with the previous version of firmware? The use of an AP without radio would be a flasher, but that is likely not what you have. Can you tell us what AP type your have?

Calimerorulez, 2024-02-25:
It is shown as SLT-EM007 Segmented AP, this one:

[image]

Running 2.07b, it's running fine:

```
env:                OpenEPaperLink_Mini_AP
build date:         2024-01-09 21:39
esp32 version:      2.07b
filesystem version: 2.07b
psram size:         2092655
flash size:         4194304
```

I just flashed it again to 2.5-beta, and now it's showing online. I don't know why. Sorry for the confusion then...

Only thing I noticed just yet, it was running channel 27, now I can't select that anymore, 26 is the highest channel.

nlimper, 2024-02-25:
Channel 27 is not supported on the C6-based AP's, but on your AP it should be displayed...
Can you please open this in your browser and paste the output here?  `http://192.168.178.198/get_ap_config` (replace the ip address for the ip-address of your AP)

jonasniesner, 2024-02-25:
I am facing the same issue with a Segmented AP. But mine has a hardware defect that something with the flashing does not work and I have to delete the AP firmware pack after every update.

nlimper, 2024-02-25:
The detection of 'no radio' is done by detecting if FLASHER_AP_RXD is shorted with FLASHER_AP_TXD.
https://github.com/jjwbruijn/OpenEPaperLink/blob/master/ESP32_AP-Flasher/src/serialap.cpp#L757-L768

Channel 27 is removed from the drop down if a C6 is detected, but I like to see the output of `/get_ap_config`  to check if that's a problem on the javascript-side or the esp32-side.

Calimerorulez, 2024-02-25:
Ok, now it's showing without radio again, after a couple of hours:

[image]

Output of `/get_ap_config`:

```
{"C6": "1", "savespace": "1", "hasFlasher": "0", "hasBLE": "0", "apstate": "7", 
  "channel": 26,
  "alias": "ap-garage",
  "led": 127,
  "tft": 255,
  "language": 0,
  "maxsleep": 0,
  "stopsleep": 1,
  "preview": 1,
  "lock": 0,
  "wifipower": 34,
  "timezone": "CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00",
  "sleeptime1": 0,
  "sleeptime2": 0,
  "ble": 0,
  "repo": "jjwbruijn/OpenEPaperLink",
  "env": "OpenEPaperLink_Mini_AP"
}
```

I manually changed it to channel 26, because the dropdown was empty, that explains.

nlimper, 2024-02-25:
Found the reason for the channel 27 not appearing: the AP always reported the presence of a C6 (which is unable to send on channel 27). Fixed in https://github.com/jjwbruijn/OpenEPaperLink/commit/b03ad07179749a7c698c62fbd1de559f0eaf8c1c
Still unknown why your AP reported to have no radio. Have you been able to reproduce, or was it a one time thing?

Calimerorulez, 2024-02-25:
I flashed 2.5-beta, and the message showed up. Reverted back to 2.07b and it disappeared. Flashed 2.5-beta again, and it shows up.

nlimper, 2024-02-25:
Okay, thanks for the info. I know enough to be able to fix it shortly.

Calimerorulez, 2024-02-25:
For what it's worth, no tags are connecting to this AP. In 2.07b there were.

I don't know if it's related.

nlimper, 2024-02-25:
fixed, https://github.com/jjwbruijn/OpenEPaperLink/commit/7e3e73a06428c85285fc045825a457831dcfb7b2
Will be visible in next release

closed 2024-02-25 by nlimper

Calimerorulez, 2024-02-26:
@nlimper 

Another thing I noticed about this AP. When it's powercycled, it 'forgets' it's channel and it reverts back to channel 11. Fixed IP address is retained, but also it's wifi network is forgotten.

nlimper, 2024-02-26:
The config is not forgotten, just not populated correctly in the web interface.
Just fixed in https://github.com/jjwbruijn/OpenEPaperLink/commit/c458c6b29ca780c8b9d6b424bbdb933f8da315b3
Forgotten wifi: for that I need more info. Can you reproduce it?

Calimerorulez, 2024-02-26:
Well, it shows channel 11 in the display of the AP too (bottom left corner).

I will try to reproduce the Wifi loss when I get back home in a couple days. I saw it because the IP address it is getting assigned was 000.000.000.000, and then it was broadcasting its accesspoint.

It's vague, I know. I will let you know if I find a way to reproduce.