OpenEPaperLink/OpenEPaperLink #237: Home Assistant images aren't displayed by the tags anymore

Calimerorulez, 2024-02-25:
Updated to 2.5-beta.

I'm using the latest Home Assistant integration (0.4.7.3). When I look in the WebUI of my accesspoint, when triggering an update from Home Assistant for the tag, I correctly see a blue 1, indicating an update.
The tag checks in, but the display of the tag isn't updated, but what the tag should display, I can now see in the WebUI (the generated image from Home Assistant).

All of my tags that are updated are behaving like that. Pushing a build in content (like current weather) is updated and displayed correctly. Only Home Assistant generated content is not displayed anymore, only in the WebUI.

Using M2 tags.

jonasniesner, 2024-02-25:
I think this belongs to the ha integration but I noticed it and will look into it.

woodshoes, 2024-02-25:
I am seeing this as well - M2's and one M3 that are not updating from HA.  
The only info I can add - is that on my M3 - a new Image gets sent to the tag - I can see it xfer successfully in the AP log.  
[image]
However, the Tag does NOT update UNTIL I press one of the buttons on the tag.  Then it is a INSTANT refresh w/ the new image.  It does not transfer again, it immediately refreshes w/ the new info. 
I hope that makes sense?  Its like the tag has the data but is waiting for something to display it
HA integration 4.7.3 - and I also took my Yellow AP to 2.5-beta but rolled it back to 2.07b after this started happening.  The tags act the same on either version of the AP.

jonasniesner, 2024-02-26:
The latest HA integration update should fix this

closed 2024-02-26 by jonasniesner

magma1447, 2024-02-27:
> The latest HA integration update should fix this

Should it be enough to upgrade the Home Assistant integration? I did that, and restarted Home Assistant, and my tags still don't update. I rebooted the AP as well, they are still not updating. I have also tried sending for example _Current date_ to one of them, but it's not receiving that data either.

magma1447, 2024-02-27:
> Should it be enough to upgrade the Home Assistant integration? I did that, and restarted Home Assistant, and my tags still don't update. I rebooted the AP as well, they are still not updating. I [...]

My issue actually seems to be something else. First off I can't send images from the AP Dashboard either. Secondly I disabled the Home assistant integration and rebooted the AP. Doesn't help. I have the issue with both _2.5 beta_ and _2.5 beta2_.

I know my last update was made a few seconds or minutes after 3 am today. Then I found this in the log:

    2024-02-27 04:00:07 Nightly reboot
    2024-02-27 04:00:12 Reboot. Reason: Software
    2024-02-27 04:00:23 000080C44C0CDADC Booting

That's about when it was supposed to update some of the tags the next time. Could it be that some scheduled nightly reboot has made the AP come into some broken state with leftover/missing files? It definitely wasn't me, I was sleeping.

What I notice is that the tagDB.json has references to jpegs in /temp, but that directory is empty. I have tried restoring a tagDB.json without success. There are jpegs named pending in the /current directory. If I start up a new tag it does show up in the Dashboard, so the radio chip is working, it's just not sending any images to any devices.

nlimper, 2024-02-27:
That's a whole different issue. Not releated to this solved issue.
There might be some bugs in the 2.5-beta/2.5-beta2, we're still investigating. You might to back to 2.07b for a few days, while we're trying to fix those.

magma1447, 2024-02-27:
> That's a whole different issue. Not releated to this solved issue. There might be some bugs in the 2.5-beta/2.5-beta2, we're still investigating. You might want to go back to 2.07b for a few days, while we're trying to fix those.

Yes, that's what I realized, that it was related. I would have created a new issue if I hadn't already commented here. I wanted to add to my comment that it wasn't the same.

I haven't managed to downgrade to 2.07b. My AP is fairly new. It came with a "custom" version (I assume pulled from git and compiled at some point in time). I can't downgrade to 2.07b, I am getting the error `No info about "OpenEPaperLink_Mini_AP_v4" found in the release.`. I interpret it as my hardware isn't supported by that release, but that could be a faulty assumption on my part.

nlimper, 2024-02-27:
Ah right, in that case you cannot downgrade. I expect a new beta very soon (tonight or tomorrow).

magma1447, 2024-02-27:
I went to https://install.openepaperlink.de/ and flashed it from there, so now it has a fresh install of 2.07c (with unknown filesystem version) instead. Same issue though.

It's not a big deal for me, it would fun to have it working though. But I'll give it a week or so and see what has been released by then. I don't have any time for this the coming two days anyway.