OpenEPaperLink/OpenEPaperLink #246: Taster nach recjts oben?

Kenaschon, 2024-02-27:
Bei den Tags mit den zwei Tastern gibt es aber noch ein kleines Problem.

Für mich macht es Sinn, das der Tag so verwendet wird, das die Taster oben sind.

Wenn ich nun die Taster aktiviere (Time Stamp) und die Displayausgabe um 180° drehe, wird die Tastenbeschreibung unten links angezeigt. Die Tasten sind aber bei 180° rechts oben.

Es geht hier drum:

[image]

Mal ein paar Fotos. So wird es glaube anschaulicher.

[image]
[image]
[image]

Grüße
Manfred

nlimper, 2024-02-27:
Good one. I will take the rotation into account. It's on my list now to fix it.

jjwbruijn, 2024-02-27:
What puzzles me, and bothers me to some lengths, is the fact that you chose to write your issue in German instead of English. I'm sure you must have noticed all the other issues are in English; even the default template (you know, the one you deleted before you typed this one up) is in English. We're trying to make this project accessible to as many people as possible, which, like it or not, involves keeping this repo in English.

Please respect this goal.

Kenaschon, 2024-02-28:
Erst mal ein nettes Hallo.

Ich finde dieses Projekt sehr spannend und interessant. Vielen Dank dafür.

Es ist aber schade, das man als "Nutzer" hier nur auf englisch eine Frage stellen darf. Nicht jeder beherrscht die englische Sprache gut genug. Diejenigen die nicht englisch sprechen, müssen also z.b. den Übersetzer von Google verwenden. Wenn ich jetzt die Frage übersetzen lasse und ich eigentlich kein englisch kann, so kann ich auch nicht sicher gehen, das meine Frage so übersetzt wurde wie ich es meine.

Wenn ich aber in meiner "Muttersprache" die Frage stelle, kann ich an der "übersetzen" Antwort eher feststellen ob meine Frage verstanden wurde. Und eins noch. Wer sich für dieses Projekt interessiert, wird wohl in der Lage sein, den Übersetzter von Google zu bedienen.

---------

First of all, a nice hello.

I find this project very exciting and interesting. Thank you for that.

But it's a shame that as a "user" you can only ask a question here in English. Not everyone knows the English language well enough. Those who don't speak English must, for example, use Google Translate. If I have the question translated and I don't actually speak English, I can't be sure that my question has been translated the way I mean it.

But if I ask the question in my "native language", I can see from the "translated" answer whether my question has been understood. And one more thing. Anyone interested in this project will probably be able to use Google Translate.

And apart from that, “Buienradar” is not necessarily an English word either. 

[image]

jjwbruijn, 2024-02-28:
See it as an opportunity to learn! People generally won’t judge you for broken English, and they’ll appreciate the fact that you try

Ilovegym66, 2024-02-28:
Hi,
thank for this excellent project ! I use the Tag with the both buttons, too, works fine. But if I change the view to 180 dregrees, the Display is changing, but the text for "Button 1" and "Button 2" are down in the screen, as before, and not up on the top, as the buttons are there... you understand the problem?

Is not only to rotate around 180 degrees, the Text with the both Buttons<No> must be changed from the bottom to the top. 
:-) 
Thanks in Advance!

closed 2024-03-18 by nlimper