OpenEPaperLink/OpenEPaperLink #247: Feature request: Track number of updates

magma1447, 2024-02-27:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Would it be feasible to track the number of updates per price tag over time? For example how many updates it has received the last 24h, the last week, and the last month?

The reason is to help understanding battery usage.

It would be a nice plus if the Home assistant integration then exposed it as a entity as well.

**Describe alternatives you've considered**
The alternative for me is to track it in Home Assistant instead. But I believe this would be a nice feature in the AP Dashboard.

nlimper, 2024-02-27:
Might be useful, yes. It will not be able to add it in the short term, but I will keep this open. Eventually, it will happen.

nlimper, 2024-03-18:
For a start, I added an simple counter to the tagdb to track the amount of updates. I'm not sure yet where to display it, but you can examine the tagDB to get the values.
I'm not planning to add sophisticated statistics.

closed 2024-03-18 by nlimper