OpenEPaperLink/OpenEPaperLink #248: Pin assignment C6

TimSchimansky, 2024-02-27:
**Describe the bug**
Not able to flash the C6 from my S3

**To Reproduce**
1. Order the gerber files from the repo (S3_C6_NanoAP)
2. Check for solder bridges
3. Flash via web flasher
4. Look into serial monitor

**Expected behavior**
The expected behavior should be a working AP, but the serial output just says:
```
I wasn't able to connect to the AP radio. Did you flash it?
The build of this firmware expects an AP tag with TXD/RXD on ESP32 pins 13 and 14, does this match with your wiring?
```
On closer inspection, I saw, that GPIO11 (S3) is connected to TX (C6) and GPIO12 (S3) is connected to RX (C6). Something is not correct here. Or maybe I have a giant twist in my head.

**Additional context**
I also build two devices, to check if i did not kill anything, but they have the exact same error type. For both of them, the webinterface is totally fine, but unusable, if there is no radio.

nlimper, 2024-02-27:
Everything works as expected. The S3 has two serial connections with the C6.
You forgot to flash the C6. For that: go to the config tab, choose update, choose 'update esp32-c6'.

TimSchimansky, 2024-02-28:
Ok if this info is part of the wiki, I somehow always skipped this while reading. It works now.