OpenEPaperLink/OpenEPaperLink #249: Issue with displays not updating after a while (new  user - might be a rooke mistake ....)

AMajland, 2024-02-28:
env: OpenEPaperLink_Mini_AP_v4
build date: 2024-02-14 10:59
esp32 version: custom <<< came with the AP

free heap: 215.32 kB ┇ free PSRAM: 7.12 MB ┇ db size: 756 bytes ┇ db record count: 4 ┇ filesystem free: 5.53 MB ┇ uptime: 3h 8m 13s

I have issue with the displays not updating after a night 
1.54" 1 from Home assistant (last updated yesterday ~21:30) . Just a test image with timestamp
1.54" 2 with Current Date (showing yesterday)
4.3 with Weather forecast

Configured for no updates between 23:00 and 5:00 - but stopped before that

[image]

The all update "last seen" but has pending updates.

The access pont has rebootet itself 3 hours ago>

but the log shows that as an expected reboot 
2024-02-28 04:00:00 Nightly reboot
2024-02-28 04:00:05 Reboot. Reason: Software
2024-02-28 04:00:10 000058364E0CDADC Booting

and HA uploads everey 15min
2024-02-28 06:45:06 http imageUpload 
2024-02-28 07:00:03 http imageUpload 

I had the same issue between monday and tuysday. I reset the displays and changed from channel 11 to channel 20 and they updated. The one from home assistant altso updated as expected everey 15min 

Should i just updata to newest beta 2.5-beta3 ?

AMajland, 2024-02-28:
Rebooting the AP did not change anything but a force update have updated the two small displys while the 4.3" is still stock

nlimper, 2024-02-28:
Yes, if you haver issues, please always first update to the latest version, in this case beta 2.5-beta3. Espcially if the release notes state "This release addresses an important issue when the 'max sleep time' is set to something bigger than 'shortest', where the tag queue would fill up, and tags become unresponsive for a while.". ;-)
Closing this, feel free to reopen if the problem is still there after updating to 2.5-beta3

closed 2024-02-28 by nlimper

magma1447, 2024-02-28:
I have the same issue (this time it really sounds like the same to me), as referenced in [issue 237](https://github.com/jjwbruijn/OpenEPaperLink/issues/237).

I updated to 2.5-beta3 this morning and none of my tags are updating with that version either.
[image]

Note that I have Home assistant data pending towards two of the tags, and a _Current date_ towards a third. For the white ones I just haven't tried to push any data after the "firmware" update this morning.