OpenEPaperLink/OpenEPaperLink #252: Cant add AP to OpenPaperlink Mini AP v4 and running in Failed state with no console information

Smurfy37, 2024-02-28:
Mini AP v4 "Failed"

**To Reproduce**
Steps to reproduce the behavior:
open web gui
look in top right corner and on ap status "Failed"
one tag has been detected however its not actually a tag i have the mac address isnt adding up and none of my epaper tags are recieving any change i apply
**Expected behavior**
Not be failed?

**Screenshots**
[image]
[image]
[image]

the logs arent showing any errors otherwise id know where to start ive reflashed ive updated the firmware etc etc nothing seems to be taking the AP out of this failed state.

Smurfy37, 2024-02-28:
Forgot to mention im running on an ESP32-S3-Wroom-1

jjwbruijn, 2024-02-28:
Can you show us what the serial console says?

peterchs, 2024-02-28:
You built it yourself?  Have you flashed the C6?  Button is at the bottom of the update firmware screen.

Smurfy37, 2024-02-28:
[image]

Smurfy37, 2024-02-28:
[image]

Smurfy37, 2024-02-28:
Just tried the C6 update and it failed i flashed using the install paperlink web flash tool

peterchs, 2024-02-28:
Says yellow AP there - did you choose the right option?
Update the filesystem too, its saying none there.
Does your AP have internet access ok to download the update?

Heres mine
[image]

Smurfy37, 2024-02-28:
ok give me two seconds ill reflash with miniAPv4 then ill check network logs see if anything getting blocked

peterchs, 2024-02-28:
Actually might have downloaded fine (if you see those files in the root of the file browser then its ok) - then its the connection to the C6 that isnt working.  Probably because its using wrong AP config/pins, try again after reflashing with the mini ap v4 fw.

nlimper, 2024-02-28:
This is a hardware defect, and out of scope of the openepaperlink project ;-)
Please send the seller (that's me...) a message, and I will help you solve it.

closed 2024-02-28 by nlimper

Smurfy37, 2024-02-28:
This is a hardware defect? i bought my esp32-s3-wroom-1 off amazon or do i need the C6 version?

nlimper, 2024-02-28:
Ah, you didn't buy a assembled Mini AP v4 on my Tindie store, but soldered it yourself? In that case, you probably don't want to use the 'OpenEpaperLink Mini AP v4' build... First identify the kind of access point you have build. Can you post a photo of your access point?

Smurfy37, 2024-02-28:
Id have loved to buy the kit but i dont trust Tindie and when i looked it seemed to be out of stock im going to donate when i get this running because i know this project wasnt easy and i appreciate the support im getting

Smurfy37, 2024-02-28:
[image]
[image]

jjwbruijn, 2024-02-28:
Okay that's an S3. You'll need both a C6 and S3 to make an accesspoint, just an S3 won't do :)

Smurfy37, 2024-02-28:
Not a problem ill keep this one and then do i attach it to a c6? or do i need just a C6? also will this work?
https://www.amazon.co.uk/Waveshare-Microcontroller-Development-Single-Core-ESP32-C6-WROOM-1-N8/dp/B0CN66P5XY/ref=sr_1_3?crid=DA271VWTFHEB&dib=eyJ2IjoiMSJ9.ZXEHJxlKe4iiGV11PVfzUIaDP7X9h8QcwTST61cq-YWaxmjqTsjn_bXfrS30WduRASr5uoZrw8d-KLCtDnil8E9VXPAcZm2ZZQr5fGTs3uJYq6sIEGXwRW8lwz3zJCx8FDWmLBruPb-GmPNPy9mYpheOdmRutQTfVlN_VmD0fvri1UPfNXaa_CptBHDfEUsrtdyoqjmWfEgAh73aO6q_J_T6PBYHcXwQCZwSYSBqTj1db5j0Xb_kK-ZzKaZkb9KTnLETLeakZp_rfnixEOcKCOW7kIq9MU1pq92rbPyo26s.JJx3bsg0F3ygCIYq5ja5VhpHMr1oDqgrlyx8x7pOcWw&dib_tag=se&keywords=ESP32-S3-c6&qid=1709144780&s=industrial&sprefix=esp32-s3-c6%2Cindustrial%2C238&sr=1-3-catcorr

Smurfy37, 2024-02-28:
bit late but can i use the Sonoff dongle with the ESP32s3 since my board has two usbs?

nlimper, 2024-02-28:
Everything is explained on the Wiki.
https://github.com/jjwbruijn/OpenEPaperLink/wiki/Building-an-Access-Point

Smurfy37, 2024-03-01:
Hi there its me again ive checked the wiki now ive got access to a C6 board and have wired as explained on the Building an Access Point Page and have not been able to get any luck

Smurfy37, 2024-03-01:
[image]
[image]

Smurfy37, 2024-03-01:
[image]

Smurfy37, 2024-03-01:
im guessing this will be the problem? do you guys have a recomended ESP32-C6 Board i can order on amazon preferably on amazon?

nlimper, 2024-03-01:
You should not try to use the web installer on the C6. Your ESP32-S3 is the main component, that can be flashed using the web installer. The C6-module is flashed from within the webinterface after installing the esp32-s3 (config tab -> update -> update C6).

Smurfy37, 2024-03-01:
[image]

nlimper, 2024-03-01:
Better! Now check your wiring between S3 and C6.
Now take a good look on your wiring, and the name of the build you did make. Hint: it's not the OpenEPaperLink_Mini_AP_v4

Smurfy37, 2024-03-01:
I followed the section on the graph for the Yellow AP / Mini AP v3 because i thought thats the closest i have to the mini ap v4 and it specifies it requires an esp32s3 so im assuming im on the right track i created this sheet so i can cut the data down to only the relevants and i can confirm the wiring is correct to the section im following? 
[image]

nlimper, 2024-03-01:
Yes, you wired according to the Yellow AP / Mini V3. But you flashed the esp32-s3 with Mini AP v4 image, which uses other pins. The 'v3' and 'v4' are different hardware revisions, not a software version. Reflash the esp32-s3 using the right build name.

Smurfy37, 2024-03-01:
[image]

reflashed with mini 3 however its installed yellow instead? and still failed?

Smurfy37, 2024-03-02:
[image]
After some pin probing I discovered that the pins 19&20 on the esp32S3 board are the wrong way round after swapping this worked perfectly thought I'd post my findings

nlimper, 2024-03-02:
Great that is works!
Yeah, with TX and RX, that's always a thing... I will add a note about that to the wiki page (and, the wiki is free to edit for everybody, so if there are more notes to add, feel free to add those)

Smurfy37, 2024-03-02:
Thanks for your help could you send me a link to donate I deffo think you guys deserve it 😁

nlimper, 2024-03-04:
Thanks! We cannot take donations, as it's too complicated to share those in a fair way among all people who are involved in the project. But we appreciate your mental support!