OpenEPaperLink/OpenEPaperLink #253: An update from version 2.07b to version 2.5-beta3 is available. (error: TypeError: Load failed)

cloudhaus, 2024-02-29:
**Describe the bug**
Receive errors when trying to update.
"error: TypeError: Load failed"

Log shows:

```
env:                ESP32_S3_16_8_YELLOW_AP
build date:         2024-01-09 15:42
esp32 version:      2.07b
filesystem version: 2.07b
psram size:         8355575
flash size:         16777216
--------------------------
Updating littleFS partition...
updating /update_actions.json
error: TypeError: Load failed
file /www/upload-demo.html.gz is up to date
updating /www/main.js.gz
error: TypeError: Load failed
updating /www/flash.js.gz
error: TypeError: Load failed
updating /www/setup.js.gz
error: TypeError: Load failed
updating /www/index.html.gz
error: TypeError: Load failed
file /www/setup.html.gz is up to date
file /www/favicon.ico.gz is up to date
updating /www/main.css.gz
error: TypeError: Load failed
file /www/variables-demo.html.gz is up to date
file /www/jsontemplate-demo.html.gz is up to date
updating /www/content_cards.json.gz
error: TypeError: Load failed
updating /www/painter.js.gz
updating /www/edit.html.gz
updating /www/ota.js.gz
updating /tag_md5_db.json
error: TypeError: Load failed
file /AP_FW_Pack.bin is up to date
updating /languages.json
error: TypeError: Load failed
updating /fonts/calibrib30.vlw
error: TypeError: Load failed
updating /fonts/twcondensed20.vlw
error: TypeError: Load failed
file /fonts/BellCent10.vlw is up to date
updating /fonts/bahnschrift30.vlw
error: TypeError: Load failed
updating /fonts/bahnschrift20.vlw
error: TypeError: Load failed
file /fonts/Signika-SB.ttf is up to date
updating /fonts/bahnschrift70.vlw
error: TypeError: Load failed
file /fonts/weathericons.ttf is up to date
file /fonts/calibrib16.vlw is up to date
file /fonts/REFSAN12.vlw is up to date
updating /tagtypes/60.json
error: TypeError: Load failed
updating /tagtypes/61.json
error: TypeError: Load failed
updating /tagtypes/62.json
file /tagtypes/E0.json is up to date
file /tagtypes/36.json is up to date
updating /tagtypes/35.json
error: TypeError: Load failed
updating /tagtypes/26.json
error: TypeError: Load failed
updating /tagtypes/B2.json
error: TypeError: Load failed
updating /tagtypes/B5.json
error: TypeError: Load failed
updating /tagtypes/27.json
error: TypeError: Load failed
updating /tagtypes/E1.json
updating /tagtypes/11.json
error: TypeError: Load failed
file /tagtypes/02.json is up to date
updating /tagtypes/BE.json
error: TypeError: Load failed
updating /tagtypes/31.json
error: TypeError: Load failed
file /tagtypes/00.json is up to date
updating /tagtypes/21.json
error: TypeError: Load failed
updating /tagtypes/05.json
error: TypeError: Load failed
updating /tagtypes/B3.json
error: TypeError: Load failed
file /tagtypes/30.json is up to date
updating /tagtypes/34.json
error: TypeError: Load failed
updating /tagtypes/2F.json
error: TypeError: Load failed
file /tagtypes/2E.json is up to date
updating /tagtypes/33.json
error: TypeError: Load failed
updating /tagtypes/01.json
error: TypeError: Load failed
file /tagtypes/F0.json is up to date
uploading /www/version.txt
------
Finished updating with 32 errors.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Update Now' in the UI

**Expected behavior**
Update to 2.5-beta3

cloudhaus, 2024-02-29:
[image]

nlimper, 2024-03-01:
Sadly I cannot reproduce it.
You can try to first update the ESP32-part, by clicking the 'esp32' button instad of 'update now'. After updating the esp32 and a reboot, you can click the 'filesystem' button, and if that works, you're fully up to date again.

cloudhaus, 2024-03-01:
Doing the updates separately worked for me. Thank you.

closed 2024-03-01 by nlimper