OpenEPaperLink/OpenEPaperLink #256: OpenEPaperLink_Mini_AP_v4 => ESP-C6 update not possible

diwoma1, 2024-03-02:
**Describe the bug**
After some time normal running of 2 new OpenEPaperLink_Mini_AP_v4, both loose the connection to the tags.
After resetting the tags, no tag can find the AP.

I have upgraded to V 2.5-Beta3, but same issue.

After reflashing to V2.07c I've tried to Update ESP-C6:
[image]

Same after update to V2.5-Beta3

**Expected behavior**
Update should work and problem with AP should disappear.

Maybe a harware-problem, because it's a new designed AP.

nlimper, 2024-03-02:
First check your AP status: 
- does the AP show 'online' in the upper right corner of the webinterface?
- what color does the neopixel on the front have?
- if you disconnect and connect a usb cable to the AP, do the two lower green LEDs flash a few times?

jjwbruijn, 2024-03-16:
Timeout waiting for reply

closed 2024-03-16 by jjwbruijn