OpenEPaperLink/OpenEPaperLink #259: Feature request:  Dayahead prices with a list of fixed surcharge

AMajland, 2024-03-03:
With DK1 added as coyntry i had to try it and it looks awesome.

[image]

But or local transport tarifs varies ower the day/night from 0,11 DKK/kWh at night to 0,43DKK/kWh when most expensive.  (and thats is just here - other regions has even bigger differences. On top of that is national system charges.

So even trough the spotprice is lowest this afternoon if was actually much cheaper last night and the aftenoon/evening prices are 

[image]

[image]

I suggest supporting a list of 24 fixed cdaily harges instead of just one.  

With 25% on top it will show the real prces

nlimper, 2024-03-03:
Apparently, those transport tarifs and national system charge also change often in DK. 
To be honest, to make such a sophisticated system is a bit out of scope for the OpenEpaperLink project. The data comes from ENTSOE, and is translated (and cached) by a Google Apps Script script. I can imagine the Goodle Apps Script to make those extra calculations, so it stays outside the OEPL project. But I prefer to have those transport tarifs fetched automatically. Is there a free API source to get them? If this goes out of hand, I will move the Google Apps Script script to a seperate github repo.

AMajland, 2024-03-03:
Yes there is a free api to https://www.energidataservice.dk where one can fetch both prices and tarifs - But is get complicated pretty fast with many different areas with their own tarif. That the reason a list of 24 values that needs to be manually updated og the ap seemed like a relatively simple solution. In most places the change again  April 1.st for summer prices.

For Home Assistant there is a good implementation See https://github.com/MTrab/energidataservice

N1 is my operator and there are more than 30 in Denmark with different prices ....

net_operator: N1
additional_tariffs:
  transmissions_nettarif: 0.0925
  systemtarif: 0.0638
  elafgift: 0.9513  <<<  Electricity tax for "normal " consumers. Since my home is heating by electricyt i only pay 0,01 after the first 4000kWh so the graph below just ignores this component
tariffs:   
  '0': 0.1376    <<< a different tarif for each hour of the day. Then add transmissions_nettarif and systemtarif on top
  '1': 0.1376
  '2': 0.1376
  '3': 0.1376
  '4': 0.1376
  '5': 0.1376
  '6': 0.4129
  '7': 0.4129
  '8': 0.4129
  '9': 0.4129
  '10': 0.4129
  '11': 0.4129
  '12': 0.4129
  '13': 0.4129
  '14': 0.4129
  '15': 0.4129
  '16': 0.4129
  '17': 1.2388
  '18': 1.2388
  '19': 1.2388
  '20': 1.2388
  '21': 0.4129
  '22': 0.4129
  '23': 0.4129
  
Since i have HA running anyway the "easy" way is probably just to draw the graphs there and send it as an image.

My prices with everything (the grey are carnot forecast)

[image]

And this is the raw spot price ( and that is what i get for excess solar)

[image]

Note the huge differences between the two graphs...

The above was just to show how it is here - and i can easily see that a full implementation is outside the scope of OEPL.

nlimper, 2024-10-28:
Done, in https://github.com/OpenEPaperLink/OpenEPaperLink/commit/0cd76eebf0214665276962b220070e7e6643161f

In the content card 'Fixed surcharge' field, you can now either enter one value (old behaviour) or an array with 24 values, for a different surcharge each hour of the day. Enter the values as `[ 0.1376, 0.1412, 0.023, ... ]`

closed 2024-10-28 by nlimper