OpenEPaperLink/OpenEPaperLink #260: Sorry, not bug, need help!

ferrexevi, 2024-03-03:
I have hanshow stellar pro tags, according to the videos on YouTube, they refer to these tags, but I can't get them to work.

I soldered the RXD, TXD, GND and VCC, which according to the schematic, are mandatory.

My intention with this is to create a NANO AP, and from that, be able to write to the other tags, which I suppose I will also have to flash, but I can't even make the AP work, except for a single tag, I also tried flashing using the Tag option Web flasher, no result.

Any ideas? suggestion? How should I proceed to get to use them?

[image]
[image]

Thanks for your time

jonasniesner, 2024-03-03:
I am sorry to tell you but you will not be able to create a Nano AP with a hanshow tag and the tag tag flasher will also not work for this tag. And you have a size that has no working drivers yet to my knowledge. Please refer to this video https://youtu.be/WRfZJ4xyYwc?si=urC03EO9kCdqfLZs on how to flash them and try to get an ESP32-C6 based AP like the yellow AP

ferrexevi, 2024-03-03:
> I am sorry to tell you but you will not be able to create a Nano AP with a hanshow tag and the tag tag flasher will also not work for this tag. And you have a size that has no working drivers yet to [...]

Yes, I saw this video, it uses only 3 pins, that confused me. I'll look at it again and see if I can do something. Maybe they are too new to be able to write in them. thank you!

nlimper, 2024-03-03:
What you have to write in them, is a piece of software. The software should of course match the hardware. It's not a question of 'too new'. But nobody created software for that tag yet. So, you cannot use it.

closed 2024-03-03 by nlimper

ferrexevi, 2024-03-03:
> What you have to write in them, is a piece of software. The software should of course match the hardware. It's not a question of 'too new'. But nobody created software for that tag yet. So, you cannot use it.

ok, we will have to wait, if my labels help someone for research to create that software, here I am.

Thank you!