OpenEPaperLink/OpenEPaperLink #261: Support for EL029H4WRC?

cgommel, 2024-03-03:
I could probably get some of the mentiones tags.

Is there any support for them in sight? 

The internals [look quite different](https://fcc.report/FCC-ID/2AFWN-EL029H4WRC/4788168.pdf
) - much smaller PCB. 

Do not have devices yet and cannot answer you what Wireless SoC is used.

jjwbruijn, 2024-03-06:
Not going to bet my life savings on it, but this tag looks Realtek, unsupported for now. But you won't know until you open it up! Tags with similar ID's have been using different SoC's anyway so based on the ID alone, it's hard to tell.

closed 2024-03-06 by jjwbruijn

BigSpice, 2024-06-29:
Got a bunch of documentation on this specific tag.
Attached below.

Also some custom work im trying to do. would love some pointers if im going about my quick pcb connections correctly.

Trying to follow https://github.com/atc1441/ATC_TLSR_Paper & https://javabin.cn/2022/epaper.html
However, I can only reset the display through my FTDi connector. 
[image]
[image]

1.6"
[image]
[image]
[image]
2.2"
[image]
[image]
[image]