OpenEPaperLink/OpenEPaperLink #267: Gicisky BLE EPD BWR 2.9" does not refresh

andorardo, 2024-03-08:
**Describe the bug**
The ESP32 AP recognizes the Gicisky 2.9", but the epaper screen doesn't refresh. Using "Force refresh" doesn't get the paper to refresh either.

**To Reproduce**
Configuration: 
* 2.52 OpenEPaperLink_Mini_AP_v4 running on S3 devkit N16R2.
* Gicisky BLE EPD BWR 2.9"
* Click on tag, change content to Static image, draw something

**Expected behavior**
epaper screen refresh with image

**Screenshots**
[image]

**Additional context**
Using https://github.com/atc1441/ATC_GICISKY_ESL, I can pair with the tag, and although it's a 2.9" device instead of the 4.2" in the demo, the tag updates (with a messed up image) and *refreshes*.

andorardo, 2024-03-08:
@atc1441 since he added the BLE capability and might know how to debug further.

atc1441, 2024-03-08:
Please insert fresh batteries in the Display 🙂

andorardo, 2024-03-08:
Voltage is back up to 3.3V :)
Sadly, that didn't quite fix it. :(

[image]

Additional info: When I enqueue a `Reboot tag` message, it just sits there, and the tag does not reboot.

Oddly enough, ATC_GICISKY_ESL does get it to refresh. Do you by any chance have a 296x128 image to try uploading with that, just to double check?

atc1441, 2024-03-08:
Please delete the tag as well from the AP and let it find it again, the Fw version is wrong displayed

andorardo, 2024-03-08:
Deleted tag a few times, rebooted AP, deleted tag again, comes up with same: `fw:33025 0x8101`

What should it be?

atc1441, 2024-03-08:
Then tha fw might be using a different upload method which leads to a bug.
Please share the uart logs from the AP.
What AP are you using and what RF interface is connected? Esp32-C6?

atc1441, 2024-03-08:
Ah sorry V4 Mini AP.
Please share then only the Uart logs in the BLE advertising section

andorardo, 2024-03-08:
Here's the relevant looking part:
```
[11:11:53]Doing the BLE Scan
[11:11:57]BLE Advertised Device found: Name: , Address: ff:ff:92:93:42:71, manufacturer data: 53503321810140, serviceUUID: 0000fef0-0000-1000-8000-00805f9b34fb, rssi: -27
[11:11:57] Payload data: 0201060302F0FE08FF53503321810140
[11:11:57] Address type: 00 Manu data: 1028CD3F070000
[11:11:57]<ADR 0033FFFF92934271
```
This recurs periodically with other RSSI values.

(do you need the other devices?)

andorardo, 2024-03-08:
There's also a bit more of this, when I do `Clear pending status`:
```
[11:21:07]BLE Image is pending
[11:21:07]BLE Could not create buffer!
[11:21:12]ping failed
[11:21:12]ping failed
[11:21:12]Resetting tag
[11:21:15]ping failed
[11:21:15]ping failed
[11:21:16]ping failed
```

nlimper, 2024-03-08:
The C6 part needs to work, or content will not be generated. If the AP showes 'failed', you should not expect it to work ;-)

andorardo, 2024-03-08:
Ah :)
How do I turn off the C6 part?

atc1441, 2024-03-08:
Exactly what @nlimper mentiones.

Otherwise the Manufacturer data looks good

The error message "BLE Could not create buffer" also hints into the problem direction.

@andorardo why can you not just flash the C6 and use it with it?

nlimper, 2024-03-08:
nope... The BLE part is very new. We didn't have a situation yet where there is no IEEE801.15.4 radio present.
If you compile it yourself: make sure that a ping just returns true, for a start. The AP will show 'online' then. It might work after that, You also might get other issues, just tackle them one by one ;-)

andorardo, 2024-03-08:
Although I have a C6 somewhere, I think this would be a good feature, so I'm trying the software route.
I hardwired sendPing to return true.

The error I get is at https://github.com/jjwbruijn/OpenEPaperLink/blob/2408b34c2b11a331841f81dc484ad9666fa8b5ae/ESP32_AP-Flasher/src/makeimage.cpp#L324

no matter if `BOARD_HAS_PSRAM` is set. 

If I uncomment `BOARD_HAS_PSRAM`, then instead I get an error at https://github.com/jjwbruijn/OpenEPaperLink/blob/2408b34c2b11a331841f81dc484ad9666fa8b5ae/ESP32_AP-Flasher/src/ble_writer.cpp#L168

presumably because now there is not enough heap.

So I made both allocations `malloc`, and reduced the BUFFER_MAX_SIZE_COMPRESSING to 20k.

Now it gets to the point of trying to connect to the BLE tag, and 
```
BLE Image is pending
BLE Compressed Length: 13620
BLE Connecting to: FF:FF:92:93:42:71
BLE onConnect
BLE starting to get service
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (BTU_TASK) 
Core  0 register dump:
PC      : 0x40383f07  PS      : 0x00060136  A0      : 0x803817f0  A1      : 0x3fcd34f0  
A2      : 0x3fccffa0  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060123  
A6      : 0x00060120  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x00060123  A11     : 0x00000000  A12     : 0x00060120  A13     : 0x00000002  
A14     : 0x02ccffa0  A15     : 0x00ffffff  SAR     : 0x00000011  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x40383f04:0x3fcd34f0 0x403817ed:0x3fcd3530 0x420b2d97:0x3fcd3570 0x420b08a2:0x3fcd35a0 0x4208d8b8:0x3fcd35d0 0x420b5049:0x3fcd3600 0x420d7b6f:0x3fcd3890 0x420b5f5a:0x3fcd3b20 0x420d7732:0x3fcd3b40 0x420b5f5a:0x3fcd3bb0 0x420d73d5:0x3fcd3bd0 0x420b53a1:0x3fcd3bf0 0x420b53cd:0x3fcd3e80 0x420b548e:0x3fcd3ea0 0x420a6f45:0x3fcd3ec0 0x420a0f22:0x3fcd4170 0x420a1462:0x3fcd4400 0x420a2169:0x3fcd4480 0x420a3ac4:0x3fcd44d0 0x420a3b69:0x3fcd44f0 0x420ac509:0x3fcd4510 0x4209e280:0x3fcd4550 0x420b2ab7:0x3fcd4570

  #0  0x40383f04:0x3fcd34f0 in compare_and_set_native at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:301
  #1  0x403817ed:0x3fcd3530 in vPortEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/include/freertos/portmacro.h:578
      (inlined by) xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:840
  #2  0x420b2d97:0x3fcd3570 in osi_thead_work_queue_put at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/thread.c:122
      (inlined by) osi_thread_post at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/thread.c:344
  #3  0x420b08a2:0x3fcd35a0 in btc_task_post at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/btc/core/btc_task.c:214
      (inlined by) btc_transfer_context at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/btc/core/btc_task.c:259
  #4  0x4208d8b8:0x3fcd35d0 in btc_gattc_cback at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gattc.c:189
  #5  0x420b5049:0x3fcd3600 in bta_gattc_search_service at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c:1133
  #6  0x420d7b6f:0x3fcd3890 in bta_gattc_search at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1546
  #7  0x420b5f5a:0x3fcd3b20 in bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:304
  #8  0x420d7732:0x3fcd3b40 in bta_gattc_disc_cmpl at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1073
  #9  0x420b5f5a:0x3fcd3bb0 in bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:304
  #10 0x420d73d5:0x3fcd3bd0 in bta_gattc_reset_discover_st at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:859
  #11 0x420b53a1:0x3fcd3bf0 in bta_gattc_explore_srvc at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c:640 (discriminator 3)
  #12 0x420b53cd:0x3fcd3e80 in bta_gattc_char_dscpt_disc_cmpl at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c:724
      (inlined by) bta_gattc_char_dscpt_disc_cmpl at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c:697
  #13 0x420b548e:0x3fcd3ea0 in bta_gattc_disc_cmpl_cback at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c:1085
  #14 0x420a6f45:0x3fcd3ec0 in gatt_end_operation at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/gatt/gatt_utils.c:2325
[... cut ...]
  #21 0x4209e280:0x3fcd4550 in btu_hci_msg_process at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/btu/btu_task.c:144
  #22 0x420b2ab7:0x3fcd4570 in osi_thread_run at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/thread.c:165
```

So some **questions** are:
1. why is `ps_malloc` failing? Does that have anything to do with the absence of the C6?
2. why is there relatively so little heap space? Does that have something to do with stuff that is not freed because of the missing C6?

nlimper, 2024-03-08:
Yes. Make sure your psram is working, as it is needed for the buffers.
In platformio.ini are some more settings about the kind of psram your board is using. The default settings assume you have 16MB flash and 8MB psram, but in your first post you stated you have 2 MB psram. That should be enough, but make sure it's detected.

andorardo, 2024-03-08:
What's the difference between the R2 and R8?

AFAICT the R2 has QD Quad SPI, which is already configured (??)
`board_build.psram_type=qspi_opi`

I'm not sure which config bit is relevant to the PSRAM -- can you give a clue?

andorardo, 2024-03-08:
Aha
`board_build.arduino.memory_type = qio_qspi`
fixed it.

For some reason it was the board_build.**arduino**.memory_type, rather than the psram_type.

Now there are no memory issues. 👍 

Things still go awry:

```
BLE Image is pending
BLE Compressed Length: 13620
BLE Connecting to: FF:FF:92:93:42:71
<ADR 000024B21AE22748
BLE onConnect
BLE starting to get service
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (BTU_TASK) 
Core  0 register dump:
PC      : 0x40383e7b  PS      : 0x00060836  A0      : 0x80381740  A1      : 0x3fcca9e0  
A2      : 0x3de078b8  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060823  
A6      : 0x00060820  A7      : 0x0000cdcd  A8      : 0x8038311e  A9      : 0xffffffff  
A10     : 0x3de078b8  A11     : 0xb33fffff  A12     : 0x3fcca9fc  A13     : 0x3fcaa755  
A14     : 0x00e078b8  A15     : 0x00ffffff  SAR     : 0x0000000f  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x40383e78:0x3fcca9e0 0x4038173d:0x3fccaa20 0x420b313f:0x3fccaa60 0x420b0c4a:0x3fccaa90 0x4208dc60:0x3fccaac0 0x420b53f1:0x3fccaaf0 0x420d7f4f:0x3fccad80 0x420b6302:0x3fccb010 0x420d7b12:0x3fccb030 0x420b6302:0x3fccb0a0 0x420d77b5:0x3fccb0c0 0x420b5749:0x3fccb0e0 0x420b5775:0x3fccb370 0x420b5836:0x3fccb390 0x420a72ed:0x3fccb3b0 0x420a12ca:0x3fccb660 0x420a180a:0x3fccb8f0 0x420a2511:0x3fccb970 0x420a3e6c:0x3fccb9c0 0x420a3f11:0x3fccb9e0 0x420ac8b1:0x3fccba00 0x4209e628:0x3fccba40 0x420b2e5f:0x3fccba60

  #0  0x40383e78:0x3fcca9e0 in spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:100
      (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:301
  #1  0x4038173d:0x3fccaa20 in vPortEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/include/freertos/portmacro.h:578
      (inlined by) xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:840
  #2  0x420b313f:0x3fccaa60 in osi_thead_work_queue_put at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/thread.c:122
      (inlined by) osi_thread_post at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/thread.c:344
  #3  0x420b0c4a:0x3fccaa90 in btc_task_post at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/btc/core/btc_task.c:214
      (inlined by) btc_transfer_context at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/btc/core/btc_task.c:259
  #4  0x4208dc60:0x3fccaac0 in btc_gattc_cback at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gattc.c:189
  #5  0x420b53f1:0x3fccaaf0 in bta_gattc_search_service at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c:1133
  #6  0x420d7f4f:0x3fccad80 in bta_gattc_search at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1546
  #7  0x420b6302:0x3fccb010 in bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:304
  #8  0x420d7b12:0x3fccb030 in bta_gattc_disc_cmpl at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1073
  #9  0x420b6302:0x3fccb0a0 in bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:304
  #10 0x420d77b5:0x3fccb0c0 in bta_gattc_reset_discover_st at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:859
  #11 0x420b5749:0x3fccb0e0 in bta_gattc_explore_srvc at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c:640 (discriminator 3)
  #12 0x420b5775:0x3fccb370 in bta_gattc_char_dscpt_disc_cmpl at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c:724
      (inlined by) bta_gattc_char_dscpt_disc_cmpl at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c:697
  #13 0x420b5836:0x3fccb390 in bta_gattc_disc_cmpl_cback at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c:1085
  #14 0x420a72ed:0x3fccb3b0 in gatt_end_operation at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/gatt/gatt_utils.c:2325
  #15 0x420a12ca:0x3fccb660 in gatt_act_discovery at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/gatt/gatt_cl.c:123
  #16 0x420a180a:0x3fccb8f0 in gatt_process_read_info_rsp at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/gatt/gatt_cl.c:499 (discriminator 4)
  #17 0x420a2511:0x3fccb970 in gatt_client_handle_server_rsp at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/gatt/gatt_cl.c:1192
  #18 0x420a3e6c:0x3fccb9c0 in gatt_data_process at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/gatt/gatt_main.c:1007
[... cut ...]
  #20 0x420ac8b1:0x3fccba00 in l2c_rcv_acl_data at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/l2cap/l2c_main.c:275
  #21 0x4209e628:0x3fccba40 in btu_hci_msg_process at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/btu/btu_task.c:144
  #22 0x420b2e5f:0x3fccba60 in osi_thread_run at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/thread.c:165
```

Somehow it never gets as far as `BLE_MAIN_STATE_UPLOAD`.

What's the source of this panic?

andorardo, 2024-03-08:
A bit of logging narrows down the error to the call to 
```c
m_haveServices = (m_semaphoreSearchCmplEvt.wait("getServices") == 0);
```
roughly here:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/adc2aee2f0d01eb6b30dd5ad3589f2cc89934beb/src/BLEClient.cpp#L435

andorardo, 2024-03-08:
This: `EXCVADDR: 0x00000000` looks like null pointer access.

andorardo, 2024-03-08:
One more thing I discovered: It is enough to simply wait 1-2 seconds (`vTaskDelay`) and the panic is triggered.

This seems to be way deep in the ESP infra.

@atc1441 , any thoughts on what might be causing this? 

Is this a bug we're tickling, or still some interaction with the AP tasks or being low on memory?

atc1441, 2024-05-06:
Perfect, and yes the issue should be fixed now

closed 2024-05-06 by atc1441

andorardo, 2024-06-03:
As of the recent 2.60 version https://github.com/OpenEPaperLink/OpenEPaperLink/releases/tag/2.60b, the new "BLE only AP" firmware Just Works. Bravo 👍