OpenEPaperLink/OpenEPaperLink #273: M3 4.2" tag firmware

LeftPinkie, 2024-03-15:
Any plans to create a firmware for the **M3 4.2"** tag (EL042H3WRA - Solum Newton with 2 buttons)?

It looks like there is already one for the 2.9" tag ([EL029H3WRA](https://github.com/jjwbruijn/OpenEPaperLink/wiki/2.9%E2%80%B3-EL029H3WRA)).

jjwbruijn, 2024-03-15:
Yeah, it’s included in the Universal firmware! Install that and it should just work.

LeftPinkie, 2024-03-15:
I still can't get the Uni firmware to flash. I have tried `SOL_M3_Uni_full_26.bin`, `SOL_M3_Uni_full_0027.bin` and `SOL_M3_Uni_ota_0027.bin`.  I am using the Mini Access Point v4 by Nic in usbflasher mode.
```
$ python3 OEPL-Flasher.py --ip 10.10.10.98 --nrf82511 --flash --exit write SOL_M3_Uni_full_26.bin
Using external port
Connected to 10.10.10.98:243
Connection with the flasher established. Version: 49
Connection established to microcontroller

Erasing flash... 
DONE!

Writing flash data from file: SOL_M3_Uni_full_26.bin

All done writing! Exiting after writing eeprom
```

After flashing I short the batteries pins and insert the batteries but all I get is bar code for the MAC. Am I doing something wrong here?

jjwbruijn, 2024-03-16:
Do you get a short green blink from the LED on booting?
Can you dump the UICR and share the results here?

LeftPinkie, 2024-03-16:
Sorry I'm a noob to microcontrollers and embedded systems... not sure how I can dump the UICR. Can I do that with the `OEPL-Flasher.py` script? I am running linux and only access to the tag is via the Mini Access Point v4.

Also I did not crack open the tag, only opened the battery panel to access the pins for flashing via Mini Access Point v4. So I cannot see any LED on-board the tag.

ferrexevi, 2024-03-16:
Is there anything new for these? Are they more modern?

[image]

Thanks

jjwbruijn, 2024-03-16:
> Sorry I'm a noob to microcontrollers and embedded systems... not sure how I can dump the UICR. Can I do that with the `OEPL-Flasher.py` script? I am running linux and only access to the tag is via [...]

Yeah, just supply -i instead of --flash and use the 'read' command to dump the UICR. It's a separate piece of flash that holds the tags config

jjwbruijn, 2024-03-16:
> Is there anything new for these? Are they more modern?
> 
> ![thumbnail_IMG_9350](https://private-user-images.githubusercontent.com/31113439/313433999-8365959c-a2f5-416b-a123-939698112d03.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA2Mjg2NDIsIm5iZiI6MTcxMDYyODM0MiwicGF0aCI6Ii8zMTExMzQzOS8zMTM0MzM5OTktODM2NTk1OWMtYTJmNS00MTZiLWExMjMtOTM5Njk4MTEyZDAzLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxNlQyMjMyMjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00OTI5Nzc2ZTFkM2RhNWEwNTM1MDk0YzdjOGQ1MjI0MDJhZjBlMGMyYjliMzFhNmI4M2U4ZWY4OTgyMDM1NDVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.HFlDFBlQYDJkayGdLi8pTHSg_bdxsRzI-mqpI1Uw8co)
> 
> Thanks

You're in luck, these tags are just getting supported!

ferrexevi, 2024-03-16:
> > Is there anything new for these? Are they more modern?
> > ![thumbnail_IMG_9350](https://private-user-images.githubusercontent.com/31113439/313433999-8365959c-a2f5-416b-a123-939698112d03.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA2Mjg2NDIsIm5iZiI6MTcxMDYyODM0MiwicGF0aCI6Ii8zMTExMzQzOS8zMTM0MzM5OTktODM2NTk1OWMtYTJmNS00MTZiLWExMjMtOTM5Njk4MTEyZDAzLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxNlQyMjMyMjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00OTI5Nzc2ZTFkM2RhNWEwNTM1MDk0YzdjOGQ1MjI0MDJhZjBlMGMyYjliMzFhNmI4M2U4ZWY4OTgyMDM1NDVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.HFlDFBlQYDJkayGdLi8pTHSg_bdxsRzI-mqpI1Uw8co)
> > Thanks
> 
> You're in luck, these tags are just getting supported! Also, please don't hijack other people's issues

Sorry, that's not my intention, I thought they were the same labels. Sorry

LeftPinkie, 2024-03-17:
> Can you dump the UICR and share the results here?

Here is dump of UICR... is it me or is the file all zero's?
[Solum_Newton_4.2-UICR.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14625615/Solum_Newton_4.2-UICR.log)

jjwbruijn, 2024-03-17:
[varuicrdump-4.2-ssd-experimental.zip](https://github.com/jjwbruijn/OpenEPaperLink/files/14625655/varuicrdump-4.2-ssd-experimental.zip)

Try writing this to the UICR!

LeftPinkie, 2024-03-17:
> Try writing this to the UICR!

Here is the dump after writing.
[Solum_Newton_4.2-UICR-2.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14625667/Solum_Newton_4.2-UICR-2.log)

jjwbruijn, 2024-03-17:
Looks good, what happens if you write the universal-full firmware to flash after writing the UICR?

LeftPinkie, 2024-03-17:
I flashed it with `SOL_M3_Uni_full_0027` and shorted the battery terminals, still nothing but the barcode of MAC.

Here is the UICR after flashing.
[Solum_Newton_4.2-UICR-2b.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14625703/Solum_Newton_4.2-UICR-2b.log)

jjwbruijn, 2024-03-17:
At least the UICR flashing went according to plan :)

Can you confirm the green led blinks on powerup?

It seems your 4.2" tag isn't compatible for whatever reason, probably a different EPD. If you have another one of this type, please try and dump the UICR and share it here, and we'll see if there's something we can do about it

LeftPinkie, 2024-03-17:
I do not get any lights from the LED on the front side... still just a barcode of the MAC. Did I brick the tag?

jjwbruijn, 2024-03-17:
Regardless of the UICR status, you should -always- get a single blink on the green LED when you boot-up the tag with a battery. Try flashing the full universal firmware again, making sure to use -f or --flash.

Do you have another one of these to dump the UICR from?

LeftPinkie, 2024-03-18:
> Do you have another one of these to dump the UICR from?

Sorry no I only have one unit to play with to see if it will work OEPL. I'll try to get another one and dump the UICR before I do any flashing. Should I dump the current firmware as well?

jjwbruijn, 2024-03-18:
Yeah if you can, please do backup the main firmware as well! Worst case, if your screen isn't supported, it'll allow us to see how the stock firmware talks to the EPD. 

As a last-ditch effort, you could try to flash v0027-full twice back to back, we've seen some tags that don't boot properly after the first flash

20cstrothman, 2024-04-06:
I actually have one of those M3 4.2" tag (EL042H3WRA). I wasn't able to get it to communicate with my flasher at all though. Actually have 2 of them, and neither responded to the flasher. Either reading or writing the EEPROM, the flasher would just say it could not establish a connection with the microcontroller. I might've had the pins connected wrong, but the exact same setup worked with some of the 2.9" M3 tags (EL029H3WR**C**). How were your pins connected? And does anyone have suggestions how to get the tag to communicate with the flasher? 
[image]
[image]
[image]

LeftPinkie, 2024-04-06:
@20cstrothman, I had to very delicately connect and manually hold the pogo pins from the flasher to the tag. Fortunately I have a flasher with a ribbon cable not the jig. On this tag, there is one missing contact pad that do not correspond with all of the pogo pins on the flasher. I don't know what this pin is for but I was able to get my flasher to communicate with the tag. If you're not getting any comm (even to read), try flipping the connection around and make sure the pogo pins are seated and lined up with pads on the tag.

ferrexevi, 2024-04-06:
> @20cstrothman, I had to very delicately connect and manually hold the pogo pins from the flasher to the tag. Fortunately I have a flasher with a ribbon cable not the jig. On this tag, there is one [...]

All information in this linck, sorry my inglish is very bad:

https://github.com/jjwbruijn/OpenEPaperLink/issues/274

I would like you to tell me how you proceeded to flash your 2.9 tags or this one that you say fails. I appreciate all the information you can give me!, thank you

20cstrothman, 2024-04-08:
@LeftPinkie, Gosh I've spent the better part of the last 3 hours trying to get this tag to work. I don't have any pogo pins and the flasher I'm using is the S2 Mini with the Tag Flasher firmware on it. I had to solder the breadboard wires onto the tag, and I've tried pretty much every way imaginable to connect the pads to the flasher. I even tried using another S2 Mini just in case the first one was having issues. Unfortunately, I still cannot get the tag to respond at all. Every time I try to read or write it says it connects to the flasher, but cannot connect to the microcontroller. I'm starting to think that the tag might be locked down and not allowing flashing of new firmware, although I cannot be sure. I'm almost to the point of shipping the tag off to someone with more expertise than me in hopes they can figure out what's going on. It certainly kinda sucks that I can't get them working right now, especially with it being so hard to get any tags whatsoever in the United States. From my research, the only place I found that I could get them was on Ebay, and the 3 listings were from Germany so shipping would be crazy (one Hanshow Stellar M3 for $17 and shipping is $32 USD). The only other place I've even seen the tags at all is at a large 'home improvement' retail store, and those aren't for sale.............. 
[image]

LeftPinkie, 2024-04-08:
@20cstrothman, I live in the US (Los Angeles) if you want to ship me the 4.2" tag and I'll try to connect with my AP & flasher. I am no expert so no guarantees that I'll get it to work or won't brick the tag (but I think it's pretty hard to brick these tags or can be easily reset if I f.u.).

LeftPinkie, 2024-04-08:
I got another Solum M2 tag to test... this time it's a Solum NEWTON LITE S-Label 2.9" [FCC ID 2AFWN-EL029H4WRC](https://fccid.io/2AFWN-EL029H4WRC).

Here are pictures...
[image]
[image]
[image]

I have tried flashing it with Universal firmware 26 & 27. After flashing I cannot boot up the device; no blink on LED.

Attached are the UICR and firmware both before and after flashing with OEPL Universal firmware.
[Solum_2AFWN-EL029H4WRC_UICR_nrf82511_ORIG_20240408_205949Z.bin.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14911796/Solum_2AFWN-EL029H4WRC_UICR_nrf82511_ORIG_20240408_205949Z.bin.log)
[Solum_2AFWN-EL029H4WRC_UICR_nrf82511_POST_20240408_215353Z.bin.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14911797/Solum_2AFWN-EL029H4WRC_UICR_nrf82511_POST_20240408_215353Z.bin.log)

[Solum_2AFWN-EL029H4WRC_flash_nrf82511_ORIG_20240408_210701Z.bin.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14911798/Solum_2AFWN-EL029H4WRC_flash_nrf82511_ORIG_20240408_210701Z.bin.log)
[Solum_2AFWN-EL029H4WRC_flash_nrf82511_POST_20240408_215509Z.bin.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14911799/Solum_2AFWN-EL029H4WRC_flash_nrf82511_POST_20240408_215509Z.bin.log)

Am I doing something wrong? Did I brick this tag? Is there such a thing on these tags and how do I recover if I did?

jjwbruijn, 2024-04-10:
That newer 4.2:
Judging from the manufacturing date, there's a decent chance the SOC is locked by Solum. While the lock is removed by a device erase, this will also erase the contents of the UICR. I don't think the current tag flasher supports removing the lock on these SOC's, as I haven't had a chance to get my hands on a locked tag. I'd give it a shot with @atc1441's SWD ESP32 flasher.

Permanently bricking a tag is nearly impossible, but judging from the fact that the led doesn't even blink, something is going wrong in the flashing process... After the full flash has been written to the soc, it should pretty much always show a single blink when booting

20cstrothman, 2024-04-11:
> I got another Solum M3 tag to test... this time it's a Solum NEWTON LITE S-Label 2.9" [FCC ID 2AFWN-EL029H4WRC](https://fccid.io/2AFWN-EL029H4WRC).
> 
> Here are pictures... ![FCC ID](https://private-user-images.githubusercontent.com/17230489/320675168-304b19b9-38b6-47d1-904e-2d3853382665.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI3OTgzODIsIm5iZiI6MTcxMjc5ODA4MiwicGF0aCI6Ii8xNzIzMDQ4OS8zMjA2NzUxNjgtMzA0YjE5YjktMzhiNi00N2QxLTkwNGUtMmQzODUzMzgyNjY1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDExVDAxMTQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAzMWE0ZjM2ZjNjMmFhZWY1YjI0ZTRlMGJiMjYwMzNjZDg3YjA0ODMyNjJkMzU0ODMyNWEwMjhhZTNjZTkxZGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pl-DOeYrPjvTef-2E5PchEhZ-3cqOj63-ibfoR1mdk4) ![front_screen](https://private-user-images.githubusercontent.com/17230489/320675171-92150249-eb0a-4e5b-9fd5-7ea18e51a8dd.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI3OTgzODIsIm5iZiI6MTcxMjc5ODA4MiwicGF0aCI6Ii8xNzIzMDQ4OS8zMjA2NzUxNzEtOTIxNTAyNDktZWIwYS00ZTViLTlmZDUtN2VhMThlNTFhOGRkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDExVDAxMTQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIyODBmMzlhY2NlMTk2NDIzNmFhZjJjZmVmNThkZmZlZGE1MDUyNzc0MmFiNGFlYjYzYTFlZDdmYjEwNDMyZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KdHl3R9EgohrOGd5U1F9vmsVWRBYwAlm2nHUrnD8rHY) ![PCB](https://private-user-images.githubusercontent.com/17230489/320675172-d52d5e47-334f-42a2-b085-0e7be6b49355.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI3OTgzODIsIm5iZiI6MTcxMjc5ODA4MiwicGF0aCI6Ii8xNzIzMDQ4OS8zMjA2NzUxNzItZDUyZDVlNDctMzM0Zi00MmEyLWIwODUtMGU3YmU2YjQ5MzU1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDExVDAxMTQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFmYjAwYTdiOWRhZTllOWQxZGY0OWJhNzg1NTBlY2QwYTg4Mzk3M2FhYTY1ZGFkMzI0OGJkODFmNWRiMGVkZWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-g4rGpDYN6LomUpZeWcP5rY587Jx7wya-3YuERKhPRw)
> 
> I have tried flashing it with Universal firmware 26 & 27. After flashing I cannot boot up the device; no blink on LED.
> 
> Attached are the UICR and firmware both before and after flashing with OEPL Universal firmware. [Solum_2AFWN-EL029H4WRC_UICR_nrf82511_ORIG_20240408_205949Z.bin.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14911796/Solum_2AFWN-EL029H4WRC_UICR_nrf82511_ORIG_20240408_205949Z.bin.log) [Solum_2AFWN-EL029H4WRC_UICR_nrf82511_POST_20240408_215353Z.bin.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14911797/Solum_2AFWN-EL029H4WRC_UICR_nrf82511_POST_20240408_215353Z.bin.log)
> 
> [Solum_2AFWN-EL029H4WRC_flash_nrf82511_ORIG_20240408_210701Z.bin.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14911798/Solum_2AFWN-EL029H4WRC_flash_nrf82511_ORIG_20240408_210701Z.bin.log) [Solum_2AFWN-EL029H4WRC_flash_nrf82511_POST_20240408_215509Z.bin.log](https://github.com/jjwbruijn/OpenEPaperLink/files/14911799/Solum_2AFWN-EL029H4WRC_flash_nrf82511_POST_20240408_215509Z.bin.log)
> 
> Am I doing something wrong? Did I brick this tag? Is there such a thing on these tags and how do I recover if I did?

That tag looks just like the 2.9 tag that I have. I was able to get it working with the SOL_M3_Uni_full_0027.bin firmware. If you look at my post in reply to ferrexevi, you'll see the pinout that I used to get it working.

olanwe, 2024-04-17:
I have an Solum Newton M3 4.2 tag (EL042H3WRA) as well, however with a BWY display.
I have flashed the SOL_M3_Uni_full_0027.bin together with varuicrdump-4.2-ssd-experimental mentioned above.
In general this worked quite well, however I have noticed two things:

1. Although I changed the MAC in the experimental UICR dump to match the tags' original MAC (using nrf_uicr_mac_patcher.py) it is different - the last two bytes do not match (Original stock-firmware MAC: 06:1c:19:79:c6:9e, now ending with ...:b6:dd). Obviously the MAC-checksum is calculated differently here...?
2. The display thinks it is BWR now, although it is in fact BWY. I tried reflashing the original UICR dump, but then the third color was not addressable any more.

P.S.: I am not sure if I should have created a separate Bug report/feature request for that - if so, please let me know and I will create a separate issue.
Original UICR dump: [EL042H3WRA_uicr.zip](https://github.com/jjwbruijn/OpenEPaperLink/files/15009722/EL042H3WRA_uicr.zip)

jjwbruijn, 2024-04-20:
@olanwe Cool, I've never had the pleasure of seeing a BWY M3 Newton. I've added detection for the BWY version UICR and pushed it to the [repo](https://github.com/jjwbruijn/OpenEPaperLink/commit/f1010ad3cc5ff17815c2a76ff7afdaefda2bf79d), would you mind checking it? Here's a completed build:
[Newton_M3_nRF52811.zip](https://github.com/jjwbruijn/OpenEPaperLink/files/15049459/Newton_M3_nRF52811.zip)

olanwe, 2024-04-21:
Thank you very much! I have uploaded it to the M3 and that looks good so far - the yellow color is back again 😀:
[image]
However, the MAC-Adress ist still different. That's totally fine with me, just wanted to note that. You can see the difference on the screenshot of the original firmware:
[image]

I also noticed that the AP is not recognizing the M3 BWY - it is displaying it as a BWR device and the standard content does not work yet (weather, date and so on), but I guess this is due to the missing tagtype defniniton on the AP.

jjwbruijn, 2024-04-21:
How about this one :)

[Newton_M3_nRF52811.zip](https://github.com/jjwbruijn/OpenEPaperLink/files/15053477/Newton_M3_nRF52811.zip)

olanwe, 2024-04-21:
Great Job, the address is now correct!
The OTA update failed this time for whatever reason and bricked the tag (with the green LED only flashing once after battery insertion), but after manual flashing everything is working perfectly again.

jonasniesner, 2024-05-06:
Can this issue be now closed?

LeftPinkie, 2024-05-06:
Yes

closed 2024-05-06 by LeftPinkie

LeftPinkie, 2024-08-22:
> I don't think the current tag flasher supports removing the lock on these SOC's, as I haven't had a chance to get my hands on a locked tag.

@jjwbruijn I can send you one of my locked SoC for you to play with.