OpenEPaperLink/OpenEPaperLink #274: New firmware SOLUM?

ferrexevi, 2024-03-16:
I have SOLUM labels, and I flashed the WEMOS S2 mini with the NANO AP option. Despite that, I still have the same error, the tag is not flashed, and in the web interface it does not detect tags, I get the same "error" as before this firmware. Are they already supposed to be compatible? What would be the procedure? Why is this not working? What am I doing wrong?

Thanks for the time and effort!

[image]
[image]
[image]
[image]

jjwbruijn, 2024-03-16:
Support for SiliconLabs-based tags is not fully there yet. It's being worked on as we speak, please be patient

jonasniesner, 2024-03-17:
Do you want to use this tag as the AP? If yes, I recommend you to get an M2 one or a C6 for this, because this one will only be supported as a tag

ferrexevi, 2024-03-17:
> Do you want to use this tag as the AP? If yes, I recommend you to get an M2 one or a C6 for this, because this one will only be supported as a tag

My goal is to be able to write to it, either as an AP or npo, then once I can write, have an AP to manage them all from one point. Right now I just want to write and I thought that for this, an AP would be needed, if not, should I use the last tag flasher option? Will I be able to write with that? The last time it failed, this time I haven't tried it.

Thanks for the reply

jjwbruijn, 2024-03-17:
> Support for SiliconLabs-based tags is not fully there yet. It's being worked on as we speak, please be patient

Please read my previous reply

ferrexevi, 2024-03-18:
> > Support for SiliconLabs-based tags is not fully there yet. It's being worked on as we speak, please be patient
> 
> Please read my previous reply

I read it, not only compatibility, it is ensuring that the connections are correct, of this model the only thing I have seen are pinouts registered as P0 25, P0 26, P027.... from the wiki of this site, they do not refer to MOSI , MISO etc...

Then the topic of tutorials, would the flashing procedure be the one mentioned? (when there is 100% compatibility). I also mentioned another model, it contains the pinout inscriptions, I hope it can help in some way.

Thank you for your comment

stevew817, 2024-07-28:
For reference - Silicon Labs firmware is now out in the open, see https://github.com/OpenEPaperLink/Tag_FW_EFR32xG22

AP firmware for these tags will likely never happen - I don't feel like doing the work as it's much easier to just build a proper spaghetti AP. Therefore closing this issue. Further discussion on Silabs tags can happen on the repo linked above.

closed 2024-07-28 by stevew817