OpenEPaperLink/OpenEPaperLink #275: Battery Voltage on display

BirkHintzeThisted, 2024-03-20:
Battery voltage 3.097 is shown as 3.97 on the device, during the waiting for data screen.

Unable to force other values to see if it's only for this specific value, but I assume all X.0XX values behave similarly.

jjwbruijn, 2024-03-20:
Thank you for this observation, can you confirm this is on an M3 tag?

BirkHintzeThisted, 2024-03-20:
That is correct. I’ll just supply all the information I have on the specific tag (even though some of it is kinda redundant).
It is the only tag I have that goes beyond 3 volts, so I don’t think it’s device specific, but probably just because it's the most likely to show the .0xx behavior.

M3 4.3” 522x152 
fw:38 0x26

jjwbruijn, 2024-03-20:
[battery voltage fix.zip](https://github.com/jjwbruijn/OpenEPaperLink/files/14666681/battery.voltage.fix.zip)
Will not make a new binary at this time for the repo, since this is a small issue. Fix will be included in future release

commit 98d9dc61 (2024-03-20): fixes #275 - battery voltage display snafu
closed 2024-03-20 by jjwbruijn