OpenEPaperLink/OpenEPaperLink #279: NRF Based M3 Tag AP firmware?

20cstrothman, 2024-03-25:
I recently got a Solum EL029H4WRC tag (N52811 chip) and was able to flash the SOL_M3_Uni_full_0027 firmware on it. It appears to work, but I do not have a tag that will work with any of the AP firmware. I was wondering if there are any plans for (or if there is any) AP firmware for the NRF based M3 tags? 
[image]
[image]
[image]
[image]

nlimper, 2024-03-25:
If you have a working OEPL access point within range, the tag will see it. No special firmware needed.
I don't know about this type, but some tag types have the zigbee antenna in the case. If you remove the pcb from the case, you don't have an antenna. If the antenna situation is solved, take a look at your AP to see if it's working at all.

closed 2024-03-25 by nlimper

atc1441, 2024-03-25:
Here is the AP Firmware for the M3 hardware: https://github.com/jjwbruijn/OpenEPaperLink/tree/master/ARM_Tag_FW/nrf52811_Platformio_AP

It will only work as an AP and will not drive the display connecred to the AP PCB

20cstrothman, 2024-03-25:
Thanks for the info. @atc1441 quick question, what all do I need to do to get the bin for the AP firmware?

atc1441, 2024-03-25:
PlatformIO

20cstrothman, 2024-03-25:
Alright, this is where I've gotten. Might you all know what I should do from here? It won't build because of this...
[image]

20cstrothman, 2024-04-08:
As an update for this, I didn't realize there were different pinouts for the flasher and the access point (duh). Once I got my computer settled and the pins connected the right way, everything worked smoothly (at least with these tags). I have some 4.2" tags, but they're not responding to the flasher for reading or writing. Also, the errors I was getting before were because I was trying to compile it on a Windows machine. I whipped up a Ubuntu VM and compiled it in there with no issues. 

Here is the pinout for the flasher
[image]

Here is the pinout for the AP
[image]