OpenEPaperLink/OpenEPaperLink #280: Configuration of all tags is lost

seeers, 2024-04-03:
As soon as I started giving all the tags a name, everything disappeared again overnight :) 
I'm not sure what exactly happened, but unfortunately all the settings (alias / content) were lost overnight. 
The RAW files from the last update are still under /current. 
I use a Yellow AP (ESP32-S3) with a C6 module as radio - Version 2.52
The content of log.txt: 

```
...
2024-04-03 03:33:44 http getJsonTemplateUrl http://openhab2/tmp/bad2.json  
2024-04-03 03:34:24 http getJsonTemplateUrl http://openhab2/tmp/bad.json  
2024-04-03 03:34:31 http getJsonTemplateUrl http://openhab2/tmp/bad2.json  
2024-04-03 03:45:08 http getJsonTemplateUrl https://script.googleusercontent.com/macros.....cencored :)   
2024-04-03 03:48:44 http getJsonTemplateUrl http://openhab2/tmp/bad2.json  
2024-04-03 03:49:25 http getJsonTemplateUrl http://openhab2/tmp/bad.json
2024-04-03 04:00:03 Nightly reboot
2024-04-03 02:00:21 Unable to connect to WiFi
2024-04-03 02:00:21 Starting configuration AP, ssid OpenEPaperLink
2024-04-03 04:00:22 Reboot. Reason: Software
2024-04-03 04:05:21 Attempting to reconnect to WiFi. 
```
[image]

Are there any similar cases?

nlimper, 2024-04-03:
Friendly reminder that a backup of the database is kept in localstorage of your browser. Using the browser you used last to config a tag, go to the config tab and choose 'restore tagDB' ('Restores the tagDB with the auto-saved version by the browser') and your tag settings are back. The tags will look like expired at first (as the 'last seen' time will be way in the past) but that solves by itself.

seeers, 2024-04-04:
Thanks for the tip, that sounds like a great function if something goes wrong. 
Unfortunately it didn't work for me, the content remains empty. 
[image]

seeers, 2024-04-04:
The whole thing is not bad for me now, are there any suspicions as to where this problem is coming from? Is there anything I can do to help localise it?

seeers, 2024-04-04:
I don't know if it helps but in my case only json templates are retrieved from a local webserver and from google. Nothing is pushed via http

foorschtbar, 2024-04-27:
Same problem here. With 2.52 it was a lot better than with 2.07b (https://github.com/jjwbruijn/OpenEPaperLink/issues/231), but still happend sometimes.

kopierschnitte, 2024-05-20:
Same here but I only pull images from http (and use the weather function). Restore always works.

closed 2024-10-28 by nlimper

foorschtbar, 2024-10-28:
@nlimper is there any fix? it happend to me every 1 or 2 weeks :(

nlimper, 2024-10-28:
I closed it due to inactivity. And I on the Discord server I don't see any recent reports that it's happening.
It's kind of impossible to debug without more detailed info/logs, as I cannot reproduce it myself. You would have to catch the UART output of the AP right when it happens, maybe that gives more info.
In any case, you can backup the database and restore it when needed, but I agree it can be annoying when you need to do that every 1-2 weeks...