OpenEPaperLink/OpenEPaperLink #285: Feature request: Add 'alignment' for "textbox" of Json template

LeftPinkie, 2024-04-06:
In the Json template, please add "alignment" attribute to the "textbox" object like it has for "text". For example:
```
{ "textbox": [x, y, w, h, "text", "fontname" }
{ "textbox": [x, y, w, h, "text", "fontname", color ] }
{ "textbox": [x, y, w, h, "text", "fontname", color, line height] }
{ "textbox": [x, y, w, h, "text", "fontname", color, line height, alignment] }
```

nlimper, 2024-11-03:
What's your usecase? IMHO, a paragraph text should (almost) always be left aligned.
For headlines etc, you can use "text" which can be left/center/right alinged.

LeftPinkie, 2024-11-03:
If I have a multi-line description, I would like to have it wrap and align center or right.

nlimper, 2024-12-10:
Done! (in https://github.com/OpenEPaperLink/OpenEPaperLink/commit/30812dff49621fc0a39c7fba8fffef922bbe345b)
`{ "textbox": [x, y, w, h, "text", "fontname", color, line height, alignment] }`
0: left, 1: center, 2: right

closed 2024-12-10 by nlimper