OpenEPaperLink/OpenEPaperLink #289: Stupid question: How to set tag content with newer FW

emufan, 2024-04-09:
**Describe the bug**
I updated from 2.01b to 2.5.2

But where can I set now the content of the tags?

[image]

If have onle the content in AP display anymore.

[image]

nlimper, 2024-04-09:
empty your browser cache, as it misses some important info in the tag types on setting the content.
You can instead also open the development window (ctrl-shift-J in Chrome), disable cache in the network tab, en reload the webpage to make sure your browser loads the latest files.

closed 2024-04-09 by nlimper

emufan, 2024-04-09:
Solved with this. Thank you!