OpenEPaperLink/OpenEPaperLink #290: Feature request: Keep set TTL

schlaggi, 2024-04-09:
**Is your feature request related to a problem? Please describe.**
I can set the TTL (in my case via Home Assistant) and it's being honored one time. If no update comes the TTL falls back to the TTL set by the AP.

**Describe the solution you'd like**
Keep the set TTL indefinitely until changed.

**Describe alternatives you've considered**
I tried rerunning the HA script before the TTL has run out, but the TTL wasn't set again. Maybe this could be an alternative solution.

nlimper, 2024-04-09:
Check your setting in the AP for 'Maximum sleep` and `Shorten latency during config'. The TTL will never be longer than the maximum sleep time, and because HA is using a websockets connection, 'Shorten latency during config' should be disabled.

schlaggi, 2024-04-09:
Thank you!
I set the maximum sleep time in the AP to 1 hour because I want the most tags to sleep as long as possible. But the tags controlled by HA should check in more often to be up-to-date which I control via the TTL in the HA script. This works but as I said only once.
I expected the tags to keep the set TTL and not fall back to the maximum sleep time.

schlaggi, 2024-04-09:
I currently use 10 minutes.

nlimper, 2024-04-09:
Sorry, '10 minutes' is not a valid value for the parameter. I sounds like nitpicking, but to be able to reproduce, and to avoid having to search for errors that are potentially not there, I like to have a clear idea about the parameters you are using. When you answer '10 minutes' I still don't know if you maybe set TTL=600 because you expect the units are in seconds, etc.

schlaggi, 2024-04-09:
Sorry here is a copy of some of the code from what I use in HA:
```
service: open_epaper_link.drawcustom
target:
  entity_id: open_epaper_link.AABBCCDDEEFF112233
data:
  background: white
  rotate: 0
  ttl: 10
  payload:
    <some text and multiline payload>
```

I initially expected the TTL to be in seconds but it obviously is in minutes which is perfectly fine.
I can see on the AP webpage that the next check in for the tag is set to 10 minutes instead of 60. But after the first ten minutes this is falling back to 60 minutes if there hasn't been an update that included the TTL _and_ forced the tag to change something on the screen.

nlimper, 2024-04-09:
Thanks, I will look in to it.

waroen, 2024-04-09:
Hi,

I can reproduce the exact same problem here, as long as there is an updated image the ttl is kept, but if the automations miss one update or they get out of sync and there isn't an update when the ttl expires it goes to the default.

I'm trying to save on the battery so I've set the maximum sleep to 1 hour. Then trying to control it, as the issue states by setting the TTL in HA.

I'd expect it to keep the same update interval as for the last image.

baschi29, 2024-04-29:
I'm also able to reproduce the issue.

However after the ttl the default it falls back to seems to be 1 minute instead of the mentioned 60 minutes. I already set max sleep to 1 hour and shorten latency during config is turned off. Is there a way to change the default behavior of the tag for the meantime? Currently it consumes a lot of battery very quickly.

nlimper, 2024-04-29:
To avoid confusion: Checking in every one minute (the default) doesn’t draw much current (if you don’t send a new image). With that, the battery is estimated 5 years life. 
The checking in time is by design: with setting the TTL during upload, you tell the AP that the image is valid for at least that time. After the time runs out, the check falls back to the default, because the tag can expect a few image shortly. If, at that point, you are sure that the image is valid for a certain time again, you can push the same image with a new TTL. ( if you push the same image, the tag will not transfer the image, and will not refresh).

closed 2024-04-29 by nlimper

baschi29, 2024-04-29:
Alright, thanks for your detailled answer!

How does the check work if one pushed the same image? By filename/metadata or by image contents?

nlimper, 2024-04-29:
The Md5 hash of the image. The last few images are kept in the storage of the tag. The AP tells the tag that there is a new image available with md5 xxxxxx. If the tag has the image in the cache, it displays the image without transferring the data.