OpenEPaperLink/OpenEPaperLink #292: Picture JSON Template

Draexl, 2024-04-10:
Hi, is it possible to add a picture in the json?

Maybe: 

[
{ "text": [5, 5, "Bahnschrift 20", "fonts/bahnschrift20", 1] },
{ "box": [10, 30, 20, 20, 2] }
{"picture: [5, 5], "/logo_test.png"}
]

nlimper, 2024-04-10:
No.
But I leave this open, maybe somebody could add it at some point.

olanwe, 2024-04-21:
As a workaround you could use [img2vlw](/olanwe/img2vlw) to convert your image into a VLW font and use it to display the image.

nlimper, 2024-12-10:
Done! (in https://github.com/OpenEPaperLink/OpenEPaperLink/commit/30812dff49621fc0a39c7fba8fffef922bbe345b)
`{ "image": [filename, x, y] }`
Places a JPG image (RGB, so no grayscale (can be tricky in some graphic software), baseline (non progressive)), located on the flash partition. The image cannot be scaled, so upload it the right size. It can be any size smaller than the tag, for example an icon. { "image": ["/testicon.jpg", 200, 50] } Tip: use an image with fixed design elements for the background, and fill in any other data using the json commands.

closed 2024-12-10 by nlimper