OpenEPaperLink/OpenEPaperLink #299: Hanshow Nebular tags (TLSR firmware) with S/N beginning 7 do not refresh display

dslatford, 2024-04-16:
Using a handful of these Hanshow tags and the alpha TLSR firmware, I've discovered the firmware works perfectly on tags with serials beginning with 3.

However while I can flash tags with serials beginning 7 and they appear in the AP, the screen never refreshes.  All tags are model Nebular-350R-N

Examining the serial output, on the working (S/N beginning 3) tags, we see the firmware report 1 second sleeps during the screen refresh, finally ending with the 40 second sleep on completion:

...
Eeprom read 4E295 Len: 256
Eeprom read 4E395 Len: 256
Eeprom read 4E495 Len: 256
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 1000 ms
Sleeping for: 40212 m

However this doesn't happen on the (newer?) tags with serials beginning with 7, the tag enters 40s sleep without the screen refreshing first:

...
Eeprom read 47295 Len: 256
Eeprom read 47395 Len: 256
Eeprom read 47495 Len: 256
Sleeping for: 40015 ms

Internally the tags appear identical. The only other visible differentiator is the working 3xxx tags all feature "D3HG" on the back while the non-working 7xxx tags all have "E3HG".

If there's any debugging I can do with the non-working tags I'm happy to assist.

Reletiv, 2024-04-19:
Send me a message on discord I'll send you a different firmware to try

nethomas1968, 2024-06-10:
I've got a few of the Nebular-350R-N units. Although I can flash the firmware in this repo, I didn't get much luck in making the actual display work. I can get serial from it, and I've made a simple command line interpreter to send commands to it. And, I can make the LED flash. But, the screen itself needs a lot more work.

dontrajik, 2024-06-14:
Use the SSD sending that usually works

bonsai308, 2024-08-05:
Unfortunately, I have the same problem. After flashing (blue LED flashes), there is no reaction from the display. It is a Nebular-350Y-N (with a 7 at the beginning of the serial number). I have tried TLSR_HS_BWY_35_ALPHA, TLSR_HS_BWY_35_SSD, TLSR_HS_BWY_35_INVERSE and TLSR_HS_BW_35_ALPHA. All without success.

dontrajik, 2024-08-05:
I made a repo for this:

https://github.com/dontrajik/Nebular

bonsai308, 2024-08-05:
> I made a repo for this:
> 
> https://github.com/dontrajik/Nebular

Thank you! Now it run

bonsai308, 2024-08-08:
Hello, it's me again. So flashing the display worked. According to the display, it is looking for the AP.
I have the Spaghetti as the AP. However, the Nebular display does not appear in the web interface. Can the Nebular even communicate with the Spaghetti AP?
The AP itself should work, since I am shown an unflashed M3 2.9 ESL.
One more off-topic question: can the ESL display also be flashed via an FTDI converter or do I have to use a "middleware" like an ESP? Thank you

nethomas1968, 2024-08-08:
I use a CH340. Probably much the same as an FTDI converter, right?