OpenEPaperLink/OpenEPaperLink #302: Add JSON render preview to demo form

olanwe, 2024-04-19:
**Is your feature request related to a problem? Please describe.**
When designing a tag image using JSON (e.g. using the demo form) you will have to send the JSON data to an exisiting tag and wait for the result to be displayed. This is somewhat annoying, because you have to wait up to 40 seconds every time plus the refresh time of the display.

**Describe the solution you'd like**
It would be great if the entered JSON could be rendered to the screen as a preview immediately, so the result can be checked right away. This might be achieved by adding a "preview online" button to the jsontemplate-demo form. Clicking this button would send the rendered JSON image to the browser window immediately (instead of transmitting it to the tag).

**Describe alternatives you've considered**
An alternative solution could be a "virtual" tag that is being rendered in a browser window instead. So you could address this virtual tag just like a regular one and see the resulting image directly after transmission. This would help checking JPEGs too and provide the possibility to check rendering on different tag sizes and colors - even those that you do not own yet.

hazbot72, 2024-04-30:
Seconded , this would be very useful

alzorin, 2024-04-30:
I would also highly appreciate this feature.

nlimper, 2024-10-28:
Both the preview and a virtual tag is a good idea. I will implement this at some point.

nlimper, 2024-11-02:
tag preview done, in https://github.com/OpenEPaperLink/OpenEPaperLink/commit/a57c797542536004cba3298d91e1668260b49106
(or, it will appear in the next release, if you're into OTA updates)

You can right click a tag and click 'tag preview', which opens a popup window with the current or pending image. The popup is refreshed in real time. So if you want to test a json, just open the popup, and upload the json. No need to wait for the refresh of a tag anymore to see the results.

closed 2024-11-02 by nlimper

alzorin, 2024-11-03:
Thank you, I am grateful. The new release came super-fast right away.

olanwe, 2024-11-05:
This is great, thank you very much! Works like a charm and simplifies testing a lot.