OpenEPaperLink/OpenEPaperLink #304: Solum 5.85 ESL

vaughnlfisher, 2024-04-22:
Not sure if this is a bug or feature request?
I have 2 versions and neither can read or write via the OEPL flasher script. I keep getting permission errors!

I can confirm that I have been successful in re flashing m3 2.9.

Attached are pictures of both.

[image]

[image]

[image]

[image]

jjwbruijn, 2024-04-22:
Doubt this is a bug :) what’s the date on the label?

vaughnlfisher, 2024-04-22:
Hi

I thought as much 😃 .
Aug 2022 and October 2022.

ferrexevi, 2024-05-04:
[image]

It is very similar to this one, this one but it is from 2023, newer does it mean that it is more complicated to write on it? Is it really feasible that something comes out to read and write on these things? What sense does it make that they take the trouble to protect it so much? ?, it would be interesting to be able to play with them in a more open way.

Great job for making this possible, even though you have to wait a while

vaughnlfisher, 2024-06-10:
Hi again

After a lot of messing around, I uploaded the following bin from another post, m3 4.2 Firmware. https://github.com/jjwbruijn/OpenEPaperLink/files/14625655/varuicrdump-4.2-ssd-experimental.zip
It has worked on a portion of the screen, and its upsdie down. But very positive.
Could you explain why this worked an none of the other FW bins I uploaded. Attached my dump
[working_infopage.zip](https://github.com/user-attachments/files/15766377/working_infopage.zip)
[image]

vaughnlfisher, 2024-06-16:
Hi did someone see this?

Misiu, 2024-09-09:
This looks like LIDL ESL (https://www.reddit.com/r/eink/comments/157iycd/seemingly_affordable_3color_eink_display_seen_at/).
Fingers crossed that these will finally be fully supported :)

MagnusErler, 2024-10-10:
I was able to flash it with `SOL_M3_Uni_full_0029.bin`

[image]

It however is not able to find my AP.
I'm running an `ESP32_S3_16_8_YELLOW_AP` AP with Bluetooth enabled. Is there anything else I need to configure to make it work? Or use a different AP?

germil, 2025-03-12:
Hello, is there anything new regarding the programming of these ESLs? I also have a few of them, specifically with the RTL8762ESL processor. How and with which firmware can I reprogram them?

👌😊

diegozalez, 2025-08-24:
Im in the same situation if anyone finds a way it will be awesome