OpenEPaperLink/OpenEPaperLink #305: Layout of hanshow nebular changed

teichtaucher, 2024-04-25:
Recently I got an 3.5" hanshow nebular pricetag. Orignally I wanted to reflash it with alternative firmware. But after some cuts into the case I found out that they changed the layout - the soldering pads for reflashing are not there as shown in this video:
https://www.youtube.com/watch?v=WRfZJ4xyYwc

[image]

There are three on the left top side of the controller. Could they be used for reflashing?

nethomas1968, 2024-06-10:
Oh, that's interesting.
I only have the other type of 3.5 inch Hanshow Nebular. The firmware in this repo doesn't actually work with this type of screen anyway, so not being able to flash it isn't going to inconvenience you.
Can you take the while thing apart and see if they have moved the 6 programming pads to some other place on the board ?

ximon, 2024-06-18:
I have the same model which I've removed from its case, here's a photo of the board:
[image]

teichtaucher, 2024-06-19:
Sorry, I missed that topic. I broke the casing of mine completely to see the contacts. It's not so easy as it was with the older revisions because the display is glued to the PCB and the battery is not a knob cell. So I was uncertain if I break everything.
Nevertheless my layout is the same as @ximon. I managed to flash the firmware and could also see the initial screen.
But there was a misunderstanding. I thought the tag works with BLE and I can control it via PC or smartphone.
This does not work and I didn't want to built up this access point. So I stopped here.
But I'm still watching OpenEpaperLink. In principle Hanshow has also BLE interface. Maybe there will be a custom firmware version which can be controlled via BLE and PC.

frr032, 2024-08-25:
Hi all. Im a total tag newbe and have had difficulty understanding what tags work and mostly where to find them. My application is displaying messages from home assistant.  I was able to get a few hanshow nebular tags that look like tags in this post from a local store. I'm hoping to flash them and use with openepapeelink. One had a broken display so I disassembled it. See pics.  I have Nic's AP/flasher. How would I flash these tags? 

[image]
[image]
[image]
[image]
[image]
[image]

teichtaucher, 2024-08-26:
You're having a model with the PIN layout which ist also shown in this video:
https://www.youtube.com/watch?v=WRfZJ4xyYwc
In this video you can also see how to flash it.
As you already cracked open you display the flash PIN connectors are shown on your second to last image.

frr032, 2024-08-27:
teichtaucher  Thank you so much for the reply!

The [link ](https://www.youtube.com/redirect?event=video_description&redir_token=QUFFLUhqa2p4SndmeTd5TV9tcGRKQll4SDVHUmh0Zk9rUXxBQ3Jtc0trWnlGSWJXdXZITzlWNWtzV0lHemxHTFJLcm1zSHNRY3pyZzF0LTVPRXQxVzVjVU94eHM5c2lQeVY3eTIycURFMmg5N0N4dDVxZWJFOGRHb0FHXzRVS3dEcWdSczRqaTlFY0c2VVY5SHRPWV9NbFk4WQ&q=https%3A%2F%2Fgithub.com%2Fjjwbruijn%2FOpenEPaperLink%2Fblob%2Fmaster%2Fbinaries%2FTag%2FTLSR_HS_35_ALPHA.bin&v=WRfZJ4xyYwc) to the tag firmware in this YouTube video above is a bad link.  I see that there are other files in it's parent folder that look similar (pic below).  

Which firmware should I use?

Also; I have the Mini-AP-v4 Access Point w/ Tag Flasher. It has a breakout board (pic below) for the pins but none of them are labeled SWS.

How should I wire this breakout board to the tags?  Is the TXD wired to the SWS pin on the tag?

Do I need to go throw the steps like you have in the video and reverse the pins to reset the tag the flash it or can I just flash it?

Thanks again with helping me get started with OPEL!  Frank

[image]

[image]

teichtaucher, 2024-08-28:
Hi, I already had a discussion in another thread regarding the Firmware for Nebular 350: https://github.com/atc1441/ATC_TLSR_Paper/issues/50#issuecomment-2308758937
You can use the info of this thread. I can remember that I used something with "350 inverted".
Regarding the MiniAP I unfortunately can't support - I stopped my Investigation before 😉

closed 2024-12-17 by atc1441

poudenes, 2026-10-14:
[image]

I found a Nebular Pro-420Q-N (M5LW)
Here the pins... Can I flash this one with a custom firmware?

Here this chip information
[image]