OpenEPaperLink/OpenEPaperLink #306: Feature request: Direct Home Assistant Zigbee integration

Migsi, 2024-04-26:
**Is your feature request related to a problem? Please describe.**
This request is not related to a real problem or similar, it more a general question to get some grasp on feasability. It is strictly related to Zigbee capable tags.

**Describe the solution you'd like**
Home assistant already provides a native way to talk to Zigbee devices. Would it be possible and feasible to get ZHA and/or Z2MQTT talk directly to the tags or are there any special quirks on how these tags use the wireless link? The basic idea would be to eliminate the need of Wifi -> Zigbee AP tags, which while quite elegant, also imply having multiple parallel Zigbee networks running. Besides that it could eliminate the need to build such a bridge tag yourself.

**Describe alternatives you've considered**
Keeping everything as is is always an option :)

atc1441, 2024-04-26:
https://github.com/jjwbruijn/OpenEPaperLink/wiki/FAQ#does-openepaperlink-operate-over-zigbee--can-i-use-my-existing-zigbee-stick-as-an-access-point

closed 2024-04-26 by atc1441

Migsi, 2024-04-26:
> https://github.com/jjwbruijn/OpenEPaperLink/wiki/FAQ#does-openepaperlink-operate-over-zigbee--can-i-use-my-existing-zigbee-stick-as-an-access-point

Pardon me, must have skipped that part when reading through it :see_no_evil: Thx for the quick answer!