OpenEPaperLink/OpenEPaperLink #307: Pin connections

LaurentN78, 2024-04-27:
Hi everybody, 
I tried to figure out which pins I from my ESP32-S3-Wroom-1 I have to connect to which pin of my ESP32-C6-Wroom 1 controller. Despite having found the "Access-point-pinouts" page, I have no clue which pins I have to connect together. Maybe anybody can help me on that issue. 
Best regards
Laurent

teichtaucher, 2024-05-02:
According to #[this](https://www.youtube.com/watch?v=_efgMQUA1F4) Video the C6 is not needed anymore.

jjwbruijn, 2024-05-02:
Yeah, that's just wrong information. The C6 is required for everything but BLE tags. Please reread the wiki page again carefully, see if that helps...

teichtaucher, 2024-05-05:
@jjwbruijn I'm a bit confused now. I flashed a Hanshow Nebular 3.5" tag with the custom firmware of atc1441. According to his video on youtube it's possible to use the ESP32 S3 only as he uses BLE for the connection to the tag. Is this not the case?
I just flashed OpenEpaperLink to my ESP32 S3 (without a C6). I just enabled bluetooth in the settings but the AP stays red:
[image]

jonasniesner, 2024-05-05:
@teichtaucher I think you got some things mixed up here. The only tags that can work with BLE only are Picksmart, Hanshow requires a C6. And you have to connect 2 pins to get the AP green at the moment.

teichtaucher, 2024-05-05:
Thanks for the answer but in these videos you can see that even hanshow communicate via BLE:
https://www.youtube.com/watch?v=geqv2qlGmOc
Furthermore just saw this:
[image]
In the FW for the Hanshow 3.5" a BLE fix was made. This would make only sense if BLE is available.

Misiu, 2024-05-06:
There is an option (hack) to build a version that doesn't require C6 - https://github.com/jjwbruijn/OpenEPaperLink/issues/267#issuecomment-2095749963
However, I agree that a standalone firmware for Bluetooth-only tags should be created.

I've been watching this project for a long time and was considering getting an AP and some tags, but they aren't available at tindie, I recently found the same video about Bluetooth tags and I'm again considering buying some tags, but if C6 or a dedicated AP isn't needed for the Bluetooth-only tags, then the entry-level would be much lower, especially that you can get C3 for less than 5$ - https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html

atc1441, 2024-05-06:
BLE is currently only working for Gicisky Displays as also mentioned in my video about that.

There is, as of now, no custom BLE firmware for Hanshow displays compatible to the OEPL BLE part. The mentioned ATC_TLSR firmware does not use a compatible BLE protocol and is a complete different project.

Anyone is of course free to ceate a Hanshow BLE firmware that is compatible to it OEPL

In general it is highly advised to build an AP with ESP32-S3 N16R8 + ESP32 C6 N8 to have the best support and compatibility!

teichtaucher, 2024-05-06:
Thanks for this clarification! This information I was looking for in the Wiki and discussions but I found no clear statement.

jonasniesner, 2024-05-06:
@Misiu even for BLE only, you will have to build and maintain your config if you want to use anything else than an ESP32-S3 N16R8 . We might make an "official" config for this esp or make it easier to use other configs like the yellow AP without a C6. This will be done to have to maintain less different configs and do less verification on our side. 

Long story short, you might save 5 bucks compared to an Esp32S3 N16R8 board but you will have to spend more time creating and testing+maintaining configs for it.

Misiu, 2024-05-06:
> Long story short, you might save 5 bucks compared to an Esp32S3 N16R8 board but you will have to spend more time creating and testing+maintaining configs for it.

Still new to all of this. I was thinking I needed ESP32-C3, but it looks like S3 will be better.
So https://www.seeedstudio.com/XIAO-ESP32S3-p-5627.html will be good or should I search for other boards?
I only plan to use the BLE Gicisky, because I don't have access to other tags and don't have the equipment to flash them.
I can reconsider buying a mini AP by NIC, but I couldn't find a good source of tags that will work with that AP (that are preflashed)

jonasniesner, 2024-05-06:
@Misiu you better search for another bord, this one only has 8MB of flash and will not be compatible without additional work. Please try to find an S3 board with 16MB of flash and 8MB of ram.

teichtaucher, 2024-05-07:
@jonasniesner Thanks for updating the Wiki yesterday. Now it's much clearer.

@LaurentN78 Sorry for taking the discussion in the wrong direction. Now I'm at the same question as you postet in your initial question. The "Access-point-pinouts" page mentioned the information. There is also an example on the bottom of the page. So assuming you are running a "Mini AP v4" you have to connect the pins listed in the esp32-s3 column to the ones in the ESP32-C6 column. E.g. PIN 47 of the esp32-s3 to EN PIN of ESP32-C6. (@all please correct me when I'm wrong)

Misiu, 2024-05-10:
I don't want to create a new topic only to ask for a board and ESL model.
Sadly I didn't find any link where to buy nanoAP (especially this one: https://github.com/jjwbruijn/OpenEPaperLink/tree/master/Hardware/S3_C6_NanoAP%20Case#variant-2)
I decided to order [ESP32S3 N16R8](https://www.aliexpress.com/item/1005006418608267.html) and [this ESL from Gicisky](https://www.aliexpress.com/item/1005002945913509.html)
according to the description, it is 2.9" 3 color (white, black, red) display. Model: PE29R_BLE.
Is anyone using this specific price tag? Does it work with OpenEPaperLink and can we use all 3 colors? On [the wiki ](https://github.com/jjwbruijn/OpenEPaperLink/wiki#ble-based-giciskypicksmart) I see info about 2.9" EPD, but I didn't found anything about that specific model

Misiu, 2024-06-09:
I'd like to share my experience with Gicisky. I've ordered 2'9 inch display, ESP32 S3 and in minutes I was able to draw custom image on the tag and integrate it with Home Assistant.
Awesome work! 🎉

tknickman, 2024-10-05:
I have all of the exact same issues described here - AP is stuck in red failed state even after enabling bluetooth. 

I was using "Mini AP v4" and an old Espressif ESP32-S3-BOX-3 ([this exact one actually](https://www.adafruit.com/product/5835)) but nothing gets it to work. 

Logs are:
> I wasn't able to connect to the AP radio. Did you flash it?
The build of this firmware expects an AP tag with TXD/RXD on ESP32 pins 18 and 17, does this match with your wiring?

I also tried the BLE firmware but that doesn't load at all and I can never get to the point of being able to configure WiFi or access the webserver. 

The logs I see there are (after a reset - that's the only time I can even get logs):
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x4214c39e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
```

Just cloned the repo today to start tweaking things but relatively new to hardware so any pointers would be great.

tknickman, 2024-10-05:
Sounds like this board might just not work because it's N16R16 instead of N16R8

tknickman, 2024-10-12:
Update - got an N16R8 ESP32 and everything works great immediately!

closed 2024-10-28 by nlimper