OpenEPaperLink/OpenEPaperLink #310: Support for Solum M3 2.7" Newton BWR - EL027H3BRA

olanwe, 2024-05-05:
**Is your feature request related to a problem? Please describe.**
I have a new M3 2.7" tag here (EL027H3BRA) and flashed SOL_M3_Uni_full_0027 on it. The device is booting up (green LED flashing, Display is refreshing), but nothing (apart from a thin colored "garbage" line on the far right of the display) is being displayed. However, the tag is polling regularly, so the firmware seems to be running normal.

[image]
[image]

Information as seen by the AP:
```
0000051D098E0006
M2 1.54"
152x152 fw:39 0x27
CH 26, RSSI -37, LQI 255, 3.163V

```
So both the tags MAC-address is inccorrect (original address is 051D098E**B490**) and both type and resolution are wrong.
Dump of UICR:
[m3_27_i_051d098eb490.bin.log](https://github.com/jjwbruijn/OpenEPaperLink/files/15212815/m3_27_i_051d098eb490.bin.log)

krikkirk, 2024-08-12:
Hi there, I dont have the same tag. You can try modifying[ this line ](https://github.com/OpenEPaperLink/Tag_FW_nRF52811/blob/main/tag_fw/hal/Newton_M3_nRF52811/tagtype_db.cpp#L113C17-L113C24)and the line below to 300 and 200. It will manually tell OEPL that this tag has the correct resolution.
And modify this [line](https://github.com/OpenEPaperLink/Tag_FW_nRF52811/blob/main/tag_fw/hal/Newton_M3_nRF52811/tagtype_db.cpp#L80) to STYPE_SIZE_026 so it will show solum 2.6 in your AP. See if the splash screen works as expected.

olanwe, 2025-05-01:
I just noticed that the tag is supported by the latest M3 firmware now, so this issue can be closed.

closed 2025-05-01 by olanwe

olanwe, 2025-05-25:
> [@olanwe](https://github.com/olanwe) which firmware did you flash to get it working? Did you also flash the UICR I have the samel problem and alrady flashed 0029 - but does not work.

Hi @michilehr, I flashed the one available here: https://github.com/OpenEPaperLink/Tag_FW_nRF52811/tree/main/tag_fw

I made a backup of the infopage first but I did not have to use it. As the tag was already running with the old firmware (i.e. logging into the AP but only displaying a thin noise line) it was even sufficient to do an over-the-air update using the OTA-version ([Newton_M3_Universal-ota.bin](https://github.com/OpenEPaperLink/Tag_FW_nRF52811/blob/main/tag_fw/Newton_M3_Universal-ota.bin)) to get it working.