OpenEPaperLink/OpenEPaperLink #316: Feature request: Support for ESP32-S3 N32R8V

rogerjak, 2024-05-11:
I got the ESP32-S3-DevKitC-1-N32R8V and tried to flash it with the Mini-AP-v4 on https://install.openepaperlink.de but it didn't work.
Then tried compiling from source, copying the V4 and changing these to all kinds of versions, but it doesn't work.

`board_build.flash_mode=dout`
`board_build.arduino.memory_type = opi_opi`
`board_build.psram_type=qspi_opi`
`board_upload.maximum_size = 16777216`
`board_upload.maximum_ram_size = 327680`
`board_upload.flash_size = 16MB`

At first I got CRC errors, but after some tweaking I got it "working", i.e. it showed up as an Wifi AP.
But in the logs I get littlefs-errors like this:

`[   650][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!`
`[  1812][E][vfs_api.cpp:105] open(): /littlefs/current/apconfig.json does not exist, no permits for creation`
`[  1982][E][Preferences.cpp:483] getString(): nvs_get_str len fail: ssid NOT_FOUND`
`[  1983][E][Preferences.cpp:483] getString(): nvs_get_str len fail: pw NOT_FOUND`
`[  2638][E][vfs_api.cpp:105] open(): /littlefs/parsers.json does not exist, no permits for creation`

Any suggestions?

nlimper, 2024-05-11:
If I remember it right, the 32MB version that you use, uses a different flash mode. The install.openepaperlink.de binaries are meant for use with the 16MB versions. But compiling it yourself should work. If you got yourself a WiFi AP, you were successfull in flashing! :-)
The errors you pasted, are to be expected with a fresh AP, and are not alarming. And sadly, `vfs_api.cpp` generates the 'does not exist' error already on a `if fs.exists(filename)` statement. So, you can safely ignore all those.

closed 2024-05-11 by nlimper

rogerjak, 2024-05-11:
I can connect to the Wifi AP in my phone, but http://192.168.4.2/ isn't working.
And nothing more in the log than what I wrote above.

nlimper, 2024-05-11:
You have to connect to the AP address, which is 192.168.4.1

rogerjak, 2024-05-11:
Ah! Stupid me, tried to connect to my phone :(
Working now, thank you!

rogerjak, 2024-05-11:
But after connecting to it and changing Wifi to my own network (working, can browse from my computer) I still get these in the log:
`[   653][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!
[  1832][E][vfs_api.cpp:105] open(): /littlefs/current/apconfig.json does not exist, no permits for creation
[  4001][E][vfs_api.cpp:105] open(): /littlefs/parsers.json does not exist, no permits for creation
[ 35268][E][vfs_api.cpp:105] open(): /littlefs/www/index.html does not exist, no permits for creation
[ 36193][E][vfs_api.cpp:105] open(): /littlefs/current/apconfig.json does not exist, no permits for creation
[ 69780][E][vfs_api.cpp:105] open(): /littlefs/current/apconfig.json does not exist, no permits for creation
[ 97412][E][vfs_api.cpp:105] open(): /littlefs/current/apconfig.json does not exist, no permits for creation
[118510][E][vfs_api.cpp:105] open(): /littlefs/current/apconfig.json does not exist, no permits for creation
[121070][E][vfs_api.cpp:105] open(): /littlefs/Tag_FW_Pack.bin does not exist, no permits for creation
[135046][E][vfs_api.cpp:105] open(): /littlefs/current/apconfig.json does not exist, no permits for creation
[139689][E][vfs_api.cpp:105] open(): /littlefs/current/apconfig.json does not exist, no permits for creation`

nlimper, 2024-05-11:
Read my first comment...
vfs_api.cpp generates the 'does not exist' error already on a if fs.exists(filename) statement. So, you can safely ignore all those.

rogerjak, 2024-05-11:
Now I can't get the C6 radio part working.
Shouldn't it be enough with the one UART (17-2, 18-3), the debug UART isn't really needed?
I get debug prints from the C6 that it receives msgs from the tags, but the AP shows nothing :(

nlimper, 2024-05-11:
All pins are needed. 
And please, with all respect, don't make this github issue a personal help channel. Please try to find the problem yourself first, read and try to interpret the serial output, and see the Wiki for documentation. The connection with the C6 is straightforward, if all pins are connected to the right pins that are mentioned in the Mini-AP-v4 build platformio.ini (and also in the wiki), it works. :-) If you finally don't get it to work, provide all relevant info for us to help you.

rogerjak, 2024-05-11:
Sorry, it started out as a request to officially support the S3 N32R8V.
But the wiki is not clear, why name same pins debug and then you tell me they're mandatory?
If I constructed this myself (software/electronics engineer, actually) it shouldn't need more than one UART for the radio.
Will order a C6 with all pins needes, so hopeflully it'll work then.
Again, sorry, won't write here no more.