OpenEPaperLink/OpenEPaperLink #319: Time Stamp - cannot be deleted

Kenaschon, 2024-05-17:
Hello everyone,

If I select Time Stamp on a TAG and then later set something else, the display remains switched when I press a button.

Can this be changed?

Thanks

Solum M3 2.9"

fw: 39 0x27

jonasniesner, 2024-05-26:
This is a bug related to preloaded images. You could reload another image for the onbuttonpressevent but I guess that does not solve your problem.

nlimper, 2024-05-26:
It should be automatically erased when you choose another content type, but that's not implemented yet.
To workaround: send CMD id 5 to the tag, this will clear all eeprom images in the tag cache: 

[image]

Kenaschon, 2024-05-26:
Thank you very much. That helped.

Best regards

Manfred