OpenEPaperLink/OpenEPaperLink #322: Yellow AP - C6 Flash troubleshooting

benv666, 2024-05-19:
Hej folks,

I've ordered components  + PCB for the Yellow AP and put one together today.

Installed firmware through https://install.openepaperlink.de/, which went fine.
Booted, connected to wifi, and tried to flash the ESP32 C6 on the module.
LEDs light up, but shortly after it states `Connection to the C6 failed`.
[image]
[image]

I've measured connectivity on the RX/TX pins and that seems ok, so not sure what the issue is.
Next step is probably to read the serial from the C6 directly, but not sure what the easiest way of doing that is -- grabbing yet another ESP32? Or is there a serial debug option somewhere in the current firmware?

Thanks for thinking along!

benv666, 2024-05-20:
Update:
Grabbed a FTDI USB dongle and hooked it up to the C6 board:
[image]

Hooked up to a serial monitor I can see a constant restart loop with seemingly empty firmware:
```
invalid header: 0xffffffff
invalid header: 0xffffffffESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x0 (SW_CLR),boot:0x9 (SPI_FAST_FLASH_BOOT)
invalid reset
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x3 (LP_SW_HPSYS),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001974a
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
```

Using some shoddy handholding of the FTDI pins I booted the bord attached to the yellow mainboard and this is what happens when the "update C6" button is pressed:
```
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid he▒ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x41 (SDIO_REI_FEO_V1_BOOT)
wait sdio download
```
after which the web interface states the now familiar
```
Flashing ESP32-C6...
Stopping AP service
C6 flash starting
Connection to the C6 failed
C6 flash end
Flashing failed. :-(
```

Not sure what it's expecting.

nlimper, 2024-05-20:
I'm not sure if that's part of the cause, but it seems you didn't place the resistor to pull GPIO8 high (footprint just next to the silkscreen 'ATC1441' text), which is needed to properly boot the C6.
Other than that, I cannot help you with the yellow pcb. If all is well, both gpio9 and the enable pin are controlled by the esp32-s3 during flashing of the c6. But anyway, if it works, it works :-)

benv666, 2024-05-20:
Thanks for that! I wasn't sure what the pad was for so I left it empty, thinking it was optional. (some more detailed build instructions would have been welcome I suppose, did some guesswork but apparently failed this one)
I'll add a 10k and see if that helps, thanks!

benv666, 2024-05-20:
Well, the resistor made no difference, still getting the same error when trying to flash the C6.
Other suggestions welcome. (or a different method of troubleshooting, I'm not sure why it's failing. Could it be a timing issue like my manual upload had?)

That said, my manually uploaded firmware works fine, now with 3 tags :D 
[image]

krikkirk, 2024-08-01:
I think you probably flipped the leds and the resistors in bottom. I see two small triangles in the drawing so likely leds should be placed in between the resistors

oywino, 2024-09-30:
Just a stupid question - what is a yellow AP and where can it be found/bought? (I googled "yellow AP" but to no avail)

nlimper, 2024-10-28:
> Just a stupid question - what is a yellow AP and where can it be found/bought? (I googled "yellow AP" but to no avail)

See https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Building-an-Access-Point

Closing this one as completed

closed 2024-10-28 by nlimper