OpenEPaperLink/OpenEPaperLink #324: Problem with tags going off-line and (perhaps related) link quality

emufan, 2024-05-21:
**Describe the bug**

Hopefully it is ok to open an issue here, because I didn't find a discussion'/help section.

Here everything run smooth until 12.05. midnight.

From there on, all of my tags have a dop in link quality. Any ideas about this? Currently auto-channel is set, one AP, andre(as always ;-) "nothing has been changed at 12.05. midnight.

More or less this is now the case on all of my tags and the change started for all of them at the same time, visible in the chart

[image]

**Main problem**

I only saw the above, as I investigated potential root causes of the main problem. Perhaps the above isn't related to the problem and perhaps the above doesn't matter at all.

The main problem is, that I have two M2 2.9" tags, which are loosing the connection in the last time very often, not at special times, but surprisingly both of them more or less at the same point in time. Both are located in different rooms, rssi is o.k. And my other M2 2.9" in same and other rooms don't have this behavior. Batteries are new and full and I have already swapped them. Only reset and re-connect are bringing them back to life.

Do you have ideas? Is there an error-log in the tag to get an idea, why this is happening for these two tags (with the same picture) coming from Home Assistant.

nlimper, 2024-05-21:
Try another channel. The default channel 11 could overlap with both WiFi and other Zigbee devices in your home. With channel 25 or 26, there is a bit less chance of conflicts. After changing it in your AP, you need to restart the tags (or wait 24 hours) for the tags to pick up the new channel id.

closed 2024-05-21 by nlimper

emufan, 2024-05-30:
@nlimper Sorry für the @ but you have already closed the topic.

I now went through all channels. And replaced the batteries. And reset several times. It stays like it is. The two tags with the same content from HA seem to hang up whyever from time to time. More ore lass always at the same point in time. And they will never get online again without reset and reconnect.

[image]

Do you have ideas? Is there an error-log in the tag to get an idea, why this is happening for these two tags (with the same picture) coming from Home Assistant. Both are in different rooms, have different rssis, etc. And the only same thing (compared to all of my other tags, which don't have the problem, is, that the same image is send to both of them.

nlimper, 2024-05-30:
When you send a new image in HA, do you specify the TTL parameter? If so, try what happens if you don't set a TTL.
tag `fw:19`, that's very old. You could start by updating the tag firmware. There is a button for it (click on the tag, then the triangle, click 'auto update', only run one tag update at a time and wait for the other one until it's finished).

emufan, 2024-05-30:
Yes,  `ttl: 300` But this is set in all service calls of the other tags as well. Even if same type with same FW.

But thanks, will try to run without and update the fw. Will get back to you with feedback anyways,

rvdv01, 2024-06-02:
I had those problems with fw19 as well.
I updated to v39 ánd replaced all batteries, so I can’t say what helped - but after that, they all have been rock solid ever since.

emufan, 2024-06-06:
Yes, direction shows really towards fw19. I have now removed all ttl and updated only one of them above. And the first time ever now only one of those (fw19) went offline and not both in parallel als always in the past. So mot probably not ttl related but fw19. Will update the other now as well.

[image]