OpenEPaperLink/OpenEPaperLink #326: Bekomme den Fehler , Cannot install jjwbruijn/OpenEPaperLink

wolfsbruder22, 2024-05-22:
Woran kann das liegen?

`$ iobroker url https://github.com/jjwbruijn/OpenEPaperLink.git --host Mini-pc-Master --debug

install jjwbruijn/OpenEPaperLink#c8e6b7b834bd772ca238531968fb7dd58e8ecf54

NPM version: 10.5.0

Installing jjwbruijn/OpenEPaperLink#c8e6b7b834bd772ca238531968fb7dd58e8ecf54... (System call)

npm WARN tarball tarball data for github:jjwbruijn/OpenEPaperLink#c8e6b7b834bd772ca238531968fb7dd58e8ecf54 (null) seems to be corrupted. Trying again.

npm WARN tarball tarball data for github:jjwbruijn/OpenEPaperLink#c8e6b7b834bd772ca238531968fb7dd58e8ecf54 (null) seems to be corrupted. Trying again.

npm ERR! code ENOENT

npm ERR! path /home/iobroker/.npm/_cacache/tmp/git-cloneZBCFcw/package.json

npm ERR! errno -2

npm ERR! enoent ENOENT: no such file or directory, open '/home/iobroker/.npm/_cacache/tmp/git-cloneZBCFcw/package.json'

npm ERR! enoent This is related to npm not being able to find a file.

npm ERR! enoent

npm ERR! A complete log of this run can be found in: /home/iobroker/.npm/_logs/2024-05-22T17_44_53_414Z-debug-0.log

npm WARN tarball tarball data for github:jjwbruijn/OpenEPaperLink#c8e6b7b834bd772ca238531968fb7dd58e8ecf54 (null) seems to be corrupted. Trying again.npm WARN tarball tarball data for github:jjwbruijn/OpenEPaperLink#c8e6b7b834bd772ca238531968fb7dd58e8ecf54 (null) seems to be corrupted. Trying again.npm ERR! code ENOENTnpm ERR! syscall opennpm ERR! path /home/iobroker/.npm/_cacache/tmp/git-cloneZBCFcw/package.jsonnpm ERR! errno -2npm ERR! enoent ENOENT: no such file or directory, open '/home/iobroker/.npm/_cacache/tmp/git-cloneZBCFcw/package.json'npm ERR! enoent This is related to npm not being able to find a file.npm ERR! enoent npm ERR! A complete log of this run can be found in: /home/iobroker/.npm/_logs/2024-05-22T17_44_53_414Z-debug-0.log

host.Mini-pc-Master Cannot install jjwbruijn/OpenEPaperLink#c8e6b7b834bd772ca238531968fb7dd58e8ecf54: 254

ERROR: Process exited with code 25`

nlimper, 2024-05-22:
Don't know what you try to accompish here, but OpenEPaperLink is not an iobroker project. Also, there is nothing to compile from the root of the project, because the repo contains multiple subprojects, for example for the esp32 side and the different tag firmwares.

closed 2024-05-22 by nlimper

slimline33, 2024-05-23:
@wolfsbruder22 

Versuch es doch mal hier: https://forum.iobroker.net/topic/66380/e-ink-display-openepaperlink-displayanzeige-mit-batterie