OpenEPaperLink/OpenEPaperLink #328: update tag type definitions- error on readme.md

rvdv01, 2024-06-02:
**Describe the bug**
Using 2.60b, when clicking’update tag type definitions’, an error is thrown

**To Reproduce**
Go to update
Click ‘update tag type definitions’

**Expected behavior**
The update runs normally

**Screenshots**
[image]

nlimper, 2024-06-02:
You can ignore the error. It would be better to skip readme.MD but it doesn’t do any harm.

rvdv01, 2024-06-02:
> You can ignore the error. It would be better to skip readme.MD but it doesn’t do any harm.

Ignoring things is what I do best :-) 

Since the error message was the last line, I was unsure if it crashed after encountering readme.md or if there were simply no files on the server that started with an ‘s’.
Also, when I retried the download, the system again said ‘updating version 0 to 1’ for every definition file, so it seemed to me some process perhaps hadn’t finished properly (eg. a process that notes the version of all definition files in some kind of database).

Thanks for the quick reply.

closed 2024-06-02 by rvdv01