OpenEPaperLink/OpenEPaperLink #335: Black distortion in yellow parts (Hansow Nebular 3.5" BWY)

NiklasCa, 2024-06-12:
Yellow color contains black dots/lines using 2.60, it was working fine using 2.52.

The phenomenon is visible in the exclamation mark in the end of atc1441's video:
https://www.youtube.com/watch?t=408&v=WRfZJ4xyYwc

I did "update tagtype definitions" since I thought it might be related to the dithering issue mentioned in the 2.60 release notes, but it made no difference.

NiklasCa, 2024-06-20:
I digged into the 2.60 changelog and realised that "yellow" is a valid color and it works when sending the content from Home Assistant, previously I used "red" and it was substituted with yellow, but that is no longer the case.
My problem is solved, my only remaining complaint would be that the "Static Image" editor (which I used only to test the display) does not support yellow, but this i no major issue for me.

nlimper, 2024-10-28:
The painter needs to be updated to use only the configured tag colors (so, no red if a tag uses yellow). To do.

nlimper, 2024-11-03:
done in https://github.com/OpenEPaperLink/OpenEPaperLink/commit/7abd5d9a84ae99253816eeea46e94e0466498615

closed 2024-11-03 by nlimper