OpenEPaperLink/OpenEPaperLink #336: 2.60b a Beta or a Final release?

AseKarlsson, 2024-06-12:
Hi,

Confused about the naming 2.60b.
Is this 2.60b a Beta or a Final release?

Thanks,
Ase

SLboat, 2024-06-29:
First time installing and using, hardware is based on yellow ip.
Used 2.6b for a few days straight, sometimes rebooting and occasionally losing connection once or twice.
Overall pretty stable.

canoben, 2024-07-04:
I only now saw your comment. I'm also on 2.60b and experience loss of connection every few hours. I've been using the AP only for a few days now.
Also, there's a bizarre comment on the firmware page saying "are you the developer?" concerning the version of the firmware.

nlimper, 2024-07-04:
2.60b is a re-release of 2.60, because of an error during initial release. You can ignore the 'b'.
'are you the developer?' is just a funny way to let you know that you run a version that is a bit newer than the latest 'official' release.
In case of problems, first check (or change) your power adapter, and check the setting 'no updates between', which is a bit buggy if you enable it. If you want to debug more, you can check the serial uart output, maybe there are clues. Also, start by minimizing the setup (so, first use it standalone and disable the HA integration) to rule out the HA integration as a cause.

closed 2024-07-04 by nlimper