OpenEPaperLink/OpenEPaperLink #338: BLE (Gicisky) size support

ptahbe, 2024-06-15:
The wiki states that the following is supported :

BLE based (Gicisky/Picksmart)
    2.13" TFT
    2.13" EPD
    2.9" EPD
    4.2" EPD

There are other sizes, like 7.5 and 10.2, available from Gicisky. Are these also supported?

jonasniesner, 2024-06-15:
Not yet but they might be supported in the future

atc1441, 2024-06-19:
@ptahbe 
I could also add them if I got my hands on one of each, but currently there is no plan to spend that money from my end

jrtraynor, 2024-06-22:
> I could also add them if I got my hands on one of each, but currently there is no plan to spend that money from my end

@atc1441   I have one of the 7.5 BLE units here. Is there anything I could do, scan/capture to help?  Can confirm the 7.5 does show up just in the oepl web page but loads as a 296x128 fw:257 0x101 , Also doesn't update the units screen.

atc1441, 2024-06-22:
You can test here: https://atc1441.github.io/ATC_GICISKY_Paper_Image_Upload.html

jrtraynor, 2024-06-22:
Got some partial success
Send CMD: 01
Got bytes: 01f400
Send CMD: 0200770100000000
Got bytes: 0200
Sending now stage 3
Send CMD: 03
Got bytes: 050000000000
Send CMD: 04
Got bytes: 0405

Using 800x480. compress,mirror and 2 colour. It does update the screen. but it comes out upside down and inverted with both the red and black txt coming out as white.  Do the same as single colour get the same just with a red/black noise background.
[image]
[image]

Not sure what to try next

atc1441, 2024-06-22:
Please get the advertising data from it

jrtraynor, 2024-06-22:
Is this what you are looking for:

[image]

atc1441, 2024-06-22:
Thanks, its at least added half way to the web uploader that way.

closed 2024-12-17 by atc1441

ZoneMR, 2025-03-29:
I too would love to see support for the 7.5" Gicisky ESL.

I'm happy to help in any way to contribute this - where do I start?