OpenEPaperLink/OpenEPaperLink #340: Gerber for S3_C6_NanoAP Variant 2 and 2.2

LeftPinkie, 2024-06-22:
Does anyone have the Gerber files for the variant 2 and 2.2 of S3_C6_NanoAP as pictured in:

https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/Hardware/S3_C6_NanoAP%20Case[/V2_Case_image_Pal_Hamre.jpg

https://github.com/OpenEPaperLink/OpenEPaperLink/blob/master/Hardware/S3_C6_NanoAP%20Case/V2_2_Case_Image_By_Viktor_Knall.png

Maybe @atc1441?

nikkisstuff, 2024-06-23:
here you go

https://github.com/OpenEPaperLink/OpenEPaperLink/tree/master/Hardware/S3_C6_NanoAP%20by%20ATC1441

closed 2024-06-25 by LeftPinkie

bjoern-r, 2024-10-18:
The gerber zip in the `S3_C6_NanoAP by ATC1441` folder only contains the square V1 variant and not the v2 variant with the 45 degree USB-C connector.