OpenEPaperLink/OpenEPaperLink #343: Help please

noha23, 2024-06-30:
I have a bunch of solum m3 price tag (EL029H3WRA) and ESP32-S3-WROOM-1 N16R8. I have flashed the esp32 as an ap and flasher. But now i am stuck because i seem not to be able to find out by myself what pin of the display connect to what pin of the esp32 to flash the custom firmware of the display. Please help
[image]

LeftPinkie, 2024-07-08:
@noha23 You can find references here: https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Access-point-pinouts. Also there is a list of access points and flashers built by other: https://github.com/OpenEPaperLink/OpenEPaperLink/wiki/Building-an-Access-Point. Hope that helps.

LeftPinkie, 2024-07-09:
Just out of curiosity, where did you get these M3 tags? Are you will to sell some to me?

BigSpice, 2024-07-15:
> Just out of curiosity, where did you get these M3 tags? Are you will to sell some to me?

I have more tags than I can handle here XD.

Sad part is I dont have an s2 based board, so im trying to find a way to use a _[WROOM-32]_ or a _[D1 MINI]_ ,to flash and create an AP for these tags. :(.

Ive also made a modified JIG for all of the pins available for the M3 2.2 Pinout.
As well as adjustments to use regular aruino cables for faster testing and debugging.

[image]

LeftPinkie, 2024-07-15:
Where are you located? I'm in the USA (Los Angeles). I would like take off your hands (buy) some of the Solum M3 tags. Also I can source and purchase the ESP-32 S3 and C6 modules for you.

BigSpice, 2024-07-16:
> Where are you located? I'm in the USA (Los Angeles). I would like take off your hands (buy) some of the Solum M3 tags. Also I can source and purchase the ESP-32 S3 and C6 modules for you.
Thank you so much :D,
Im here in Ottawa ON.
Send me an email, im fully able to send you some.
jordan.k.m@proton.me

closed 2024-09-04 by nlimper