OpenEPaperLink/OpenEPaperLink #349: FTP server for Access Point FW

LeftPinkie, 2024-07-07:
Request to add FTP server to access point fw. This will allow us to upload & manage files on the AP. While the edit page (`http://{ap_ip_address}/edit`) is adequate to manage one file at a time, FTP would allow us to upload, delete, download multiple files. Also, please allow ability to create/delete directories.

jonasniesner, 2024-07-15:
I have to close this as not planed for now. The ESP is at its limits at the moment and adding FTP would most likely push it over. We maybe can think about it after S2 based APs are discontinued.

closed 2024-07-15 by jonasniesner

LeftPinkie, 2024-07-15:
Maybe include in the AP binary but not activated / enabled by default... and have it as an option to enable via the admin page